realthunder/FreeCAD #97: Error. The solver shows an error in constraint 102, which is missing from the list of constraints.

tiket18, 2021-02-02:
Error. The solver shows an error in constraint 102, which is missing from the list of constraints.
[image]

realthunder, 2021-02-03:
Above the constraint list view in the task panel, you can uncheck `Hide internal alignment` option to reveal the constraint 102. But since it is an internal constraint, it can't be deleted, I don't know how this can be fixed. Maybe let's ask @abdullahtahiriyo, any idea how to deal with redundant internal constraint?

abdullahtahiriyo, 2021-04-07:
@realthunder @tiket18 
An internal alignment constraint can be redundant with other constraints, but it is never the one to be removed. 
It may be possible to improve the algorithm suggesting redundant constraints, so that it does not suggest removing internal alignment constraints.

If you still have the file, it may be helpful if this were handled in the help forum:
https://forum.freecadweb.org/viewforum.php?f=3

I am creating a bug report now to revisit this topic.

Thanks for reporting! :)

abdullahtahiriyo, 2021-04-07:
https://tracker.freecadweb.org/view.php?id=4619

luzpaz, 2022-01-19:
@tiket18 can you please post an example file

tiket18, 2022-01-19:
That's not what we're talking about here at all.The general list of constraints is shown as a list at the bottom.  The solver prompts you to delete a constraint that is not on the list. We can't play in the dark, can we? I'm supposed to understand which restriction is suggested to be removed?
The problem persists in the current build.

dickshaydle, 2022-01-19:
@tiket18 Which version are you using? Looks like an older one.
It is a good habit to always include about-version info when you are filing an issue.

tiket18, 2022-01-20:
We are discussing a ticket from a year and a half ago. And it has been reproduced all this time, in different versions and builds.
There's no need to "батон крошить" on the realthunder build that I mostly use - the respected Zheng Lei has no effect on the solver.

dickshaydle, 2022-01-20:
> Above the constraint list view in the task panel, you can uncheck `Hide internal alignment` option to reveal the constraint 102.

> An internal alignment constraint can be redundant with other constraints, but it is never the one to be removed.

> @tiket18 can you please post an example file

Still think it would be helpful if you post an example file of the issue. Preferably a minimal example where you delete all constraints that don't cause the issue.