realthunder/FreeCAD #98: Tiny area to switch between open files

sigxcpu76, 2021-02-03:
Well, I've managed to reproduce this thing I've mentioned some time ago.
Sometimes, the area where you can click in the opened files toolbar is very narrow.
I've attached a clip that shows this behavior.

https://user-images.githubusercontent.com/737085/106679033-84be6e00-65c4-11eb-9ffb-61d76f658d4a.mp4

realthunder, 2021-02-03:
Do you have any overlay dockable at the bottom?

sigxcpu76, 2021-02-03:
They were all moved to the right.

[image]

luzpaz, 2022-01-19:
Can you still reproduce this @sigxcpu76

sigxcpu76, 2022-01-19:
I didn't notice it for the past few versions so I guess it is fixed. Sorry for not coming back to this.

closed 2022-01-19 by sigxcpu76