realthunder/FreeCAD #99: Small build issue [Archlinux]

Salamandar, 2021-02-04:
Hi, 
I'm the packager of freecad-linkstage3-git on AUR, and I lately had a build issue : (sorry for the french)

```
../src/Gui/WidgetFactory.cpp: Dans la fonction membre « Py::Object Gui::PythonWrapper::fromQEvent(QEvent*, const char*) »:
../src/Gui/WidgetFactory.cpp:794:37: erreur: « object » n'a pas été déclaré dans cette portée; vouliez-vous employer « _object » ?
  794 |     return qt_wrapInstance<QEvent*>(object, className, "shiboken2", "PySide2.QtCore", "wrapInstance");
      |                                     ^~~~~~
      |                                     _object
```

Salamandar, 2021-02-07:
Looks like it fixes the issue. Thanks !!

ceremcem, 2021-02-25:
@Salamandar Forgot to close the issue?

Salamandar, 2021-02-25:
Yeah. Closing.

closed 2021-02-25 by Salamandar