realthunder/FreeCAD #102: Expression editor appears to float with no window background?

brandonbeaumia, 2021-02-07:
Just installed the most recently daily (2021-02-05) and noticed this issue. I've tried changing my style sheets and the problem persisted.

[image]

realthunder, 2021-02-07:
This is a feature actually. You can move and resize this edgeless dialog just the same by moving your mouse around the edge or top area. If you want the original normal dialog, uncheck `Preference -> Display -> UI -> Expression Editor -> In place edit`.

brandonbeaumia, 2021-02-07:
OH! The look was so jarring I assumed there was something going wrong. I like that it pops up nearby, but the floating boxes look incredibly awkward. Glad that I can simply turn the feature off, thank you for pointing me in the right direction!

closed 2022-01-19 by luzpaz