realthunder/FreeCAD #104: sketches break down.

tiket18, 2021-02-14:
[image]
[image]

In versions older than freecad-linkstage3-git-asm3.0.11.r4124.gd7b79274aa-1, thumbnails break when opened.

Here's a screenshot (the file itself - by mail), sketch022, on the left in freecad-linkstage3-git-asm3.0.11.r4124.gd7b79274aa-1, on the right in freecad-linkstage3-git-asm3.0.11.r4498.g08fdaf6a5f-1-x86_64.pkg.tar.xz

realthunder, 2021-02-20:
Are you compiling you own binary? With the current commit, I can select the redundant constraint and delete it. There are quite a few redundant ones, and you'll have to repeat the deleting process until they are all gone. I guess it is cause by changes in sketch solver.

tiket18, 2021-02-20:
That's what I'm talking about = there's a problem in the sketch solver. And this problem is new, it wasn't there before.

luzpaz, 2022-01-19:
@realthunder anything to do here?