realthunder/FreeCAD #105: LinkStage3 crashes when saving file when using Sketcher workbench

christian-nils, 2021-02-15:
Hi!

I am using LinkStage3 built at https://github.com/realthunder/FreeCAD/commit/26e7f6394d9c1152602ddb3978b207242602c70c.
I am using https://github.com/realthunder/coin/commit/5dd4b9854bc4d5c8bbd89093204c7702d8dcc99f and my Qt Version is 5.15.2.

When I save my file when creating a sketch, FreeCAD crashes. Here is a video showing the process (https://youtu.be/WSR2lZ4lMGk), and here is the message error:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib64/libc.so.6(+0x3da60) [0x7f46efcc7a60]
#1  0x7f46f1c4b875 in SoAction::traverse(SoNode*) from /usr/local/lib64/libCoin.so.80+0x15
#2  0x7f46f55f738a in Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) from /usr/local/lib64/libFreeCADGui.so+0x238
#3  0x7f46f1c4c481 in SoAction::apply(SoPathList const&, int) from /usr/local/lib64/libCoin.so.80+0x541
#4  0x7f46f55f6cd1 in Gui::SoBoxSelectionRenderActionP::apply(Gui::SoBoxSelectionRenderAction*, SoPathList const&, int) from /usr/local/lib64/libFreeCADGui.so+0x195
#5  0x7f46f55f6d25 in Gui::SoBoxSelectionRenderAction::apply(SoPathList const&, int) from /usr/local/lib64/libFreeCADGui.so+0x31
#6  0x7f46f1c52035 in SoGLRenderActionP::renderSingle(SoNode*) from /usr/local/lib64/libCoin.so.80+0x245
#7  0x7f46f1c524df in SoGLRenderActionP::render(SoNode*) from /usr/local/lib64/libCoin.so.80+0xff
#8  0x7f46f55f738a in Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) from /usr/local/lib64/libFreeCADGui.so+0x238
#9  0x7f46f1c4bd9a in SoAction::apply(SoNode*) from /usr/local/lib64/libCoin.so.80+0x2aa
#10  0x7f46f55f621a in Gui::SoBoxSelectionRenderAction::apply(SoNode*) from /usr/local/lib64/libFreeCADGui.so+0x32
#11  0x7f46f56f1b18 in Gui::View3DInventorViewer::renderToFramebuffer(QOpenGLFramebufferObject*) from /usr/local/lib64/libFreeCADGui.so+0x2b4
#12  0x7f46f56f14e8 in Gui::View3DInventorViewer::imageFromFramebuffer(int, int, int, QColor const&, QImage&) from /usr/local/lib64/libFreeCADGui.so+0x1ea
#13  0x7f46f527a15b in Gui::Thumbnail::SaveDocFile(Base::Writer&) const from /usr/local/lib64/libFreeCADGui.so+0x16b
#14  0x7f46f2c07462 in Base::ZipWriter::writeFiles() from /usr/local/lib64/libFreeCADBase.so+0xd2
#15  0x7f46f38dd25d in App::Document::save(Base::Writer&, bool) const from /usr/local/lib64/libFreeCADApp.so+0x1d1
#16  0x7f46f38dc467 in App::Document::saveToFile(char const*) const from /usr/local/lib64/libFreeCADApp.so+0x43d
#17  0x7f46f38dbfbc in App::Document::save() from /usr/local/lib64/libFreeCADApp.so+0x26a
#18  0x7f46f3a6b470 in App::DocumentPy::save(_object*) from /usr/local/lib64/libFreeCADApp.so+0x50
#19  0x7f46f3a5f3ed in App::DocumentPy::staticCallback_save(_object*, _object*) from /usr/local/lib64/libFreeCADApp.so+0xfb
#20  /usr/lib64/libpython3.9.so.1.0(+0x12292a) [0x7f46f237e92a]
#21  /usr/lib64/libpython3.9.so.1.0(_PyObject_MakeTpCall+0x83) [0x7f46f236f8f3]
#22  /usr/lib64/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x5a38) [0x7f46f236c558]
#23  /usr/lib64/libpython3.9.so.1.0(+0x109bd1) [0x7f46f2365bd1]
#24  /usr/lib64/libpython3.9.so.1.0(_PyEval_EvalCodeWithName+0x55) [0x7f46f23e4435]
#25  /usr/lib64/libpython3.9.so.1.0(PyEval_EvalCodeEx+0x3d) [0x7f46f23e43cd]
#26  /usr/lib64/libpython3.9.so.1.0(PyEval_EvalCode+0x1f) [0x7f46f23e437f]
#27  /usr/lib64/libpython3.9.so.1.0(+0x1b523d) [0x7f46f241123d]
#28  /usr/lib64/libpython3.9.so.1.0(+0x1b0c26) [0x7f46f240cc26]
#29  /usr/lib64/libpython3.9.so.1.0(PyRun_StringFlags+0xa8) [0x7f46f24053b8]
#30  0x7f46f2b66d16 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/local/lib64/libFreeCADBase.so+0xd2
#31  0x7f46f52c1e62 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /usr/local/lib64/libFreeCADGui.so+0xca
#32  0x7f46f52c1b37 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /usr/local/lib64/libFreeCADGui.so+0xfd
#33  0x7f46f51abe21 in Gui::Document::save() from /usr/local/lib64/libFreeCADGui.so+0x56b
#34  0x7f46f5885f7d in Gui::MDIView::onMsg(char const*, char const**) from /usr/local/lib64/libFreeCADGui.so+0xe3
#35  0x7f46f56dd4cd in Gui::View3DInventor::onMsg(char const*, char const**) from /usr/local/lib64/libFreeCADGui.so+0xb1b
#36  0x7f46f5115a37 in Gui::Application::sendMsgToActiveView(char const*, char const**) from /usr/local/lib64/libFreeCADGui.so+0x51
#37  0x7f46f517afbd in Gui::Application::sSendActiveView(_object*, _object*) from /usr/local/lib64/libFreeCADGui.so+0x7d
#38  /usr/lib64/libpython3.9.so.1.0(+0x12292a) [0x7f46f237e92a]
#39  /usr/lib64/libpython3.9.so.1.0(_PyObject_MakeTpCall+0x83) [0x7f46f236f8f3]
#40  /usr/lib64/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x5a38) [0x7f46f236c558]
#41  /usr/lib64/libpython3.9.so.1.0(+0x109bd1) [0x7f46f2365bd1]
#42  /usr/lib64/libpython3.9.so.1.0(_PyEval_EvalCodeWithName+0x55) [0x7f46f23e4435]
#43  /usr/lib64/libpython3.9.so.1.0(PyEval_EvalCodeEx+0x3d) [0x7f46f23e43cd]
#44  /usr/lib64/libpython3.9.so.1.0(PyEval_EvalCode+0x1f) [0x7f46f23e437f]
#45  /usr/lib64/libpython3.9.so.1.0(+0x1b523d) [0x7f46f241123d]
#46  /usr/lib64/libpython3.9.so.1.0(+0x1b0c26) [0x7f46f240cc26]
#47  /usr/lib64/libpython3.9.so.1.0(PyRun_StringFlags+0xa8) [0x7f46f24053b8]
#48  0x7f46f2b66d16 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/local/lib64/libFreeCADBase.so+0xd2
#49  0x7f46f52c1e62 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /usr/local/lib64/libFreeCADGui.so+0xca
#50  0x7f46f52c1b37 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /usr/local/lib64/libFreeCADGui.so+0xfd
#51  0x7f46f52d8963 in StdCmdSave::activated(int) from /usr/local/lib64/libFreeCADGui.so+0x31
#52  0x7f46f52c0ae4 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/local/lib64/libFreeCADGui.so+0x3b6
#53  0x7f46f528fd40 in Gui::Action::onActivated() from /usr/local/lib64/libFreeCADGui.so+0x26
#54  /usr/local/lib64/libFreeCADGui.so(+0x11f83db) [0x7f46f529a3db]
#55  /usr/lib64/libQt5Core.so.5(+0x2b13c0) [0x7f46f055f3c0]
#56  0x7f46f0fbe646 in QAction::triggered(bool) from /usr/lib64/libQt5Widgets.so.5+0x46
#57  0x7f46f0fc0f31 in QAction::activate(QAction::ActionEvent) from /usr/lib64/libQt5Widgets.so.5+0xe1
#58  /usr/lib64/libQt5Widgets.so.5(+0x29f732) [0x7f46f10b6732]
#59  0x7f46f10b6855 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib64/libQt5Widgets.so.5+0xf5
#60  0x7f46f11aa43e in QToolButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib64/libQt5Widgets.so.5+0xe
#61  0x7f46f1005b1e in QWidget::event(QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x1ce
#62  0x7f46f0fc4ec3 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x83
#63  0x7f46f0fcbeeb in QApplication::notify(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x4ab
#64  0x7f46f525f56d in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/local/lib64/libFreeCADGui.so+0xff
#65  0x7f46f052fbd8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib64/libQt5Core.so.5+0x128
#66  0x7f46f0fcaefa in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib64/libQt5Widgets.so.5+0x1da
#67  /usr/lib64/libQt5Widgets.so.5(+0x207375) [0x7f46f101e375]
#68  /usr/lib64/libQt5Widgets.so.5(+0x20a6be) [0x7f46f10216be]
#69  0x7f46f0fc4ec3 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x83
#70  0x7f46f525f56d in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/local/lib64/libFreeCADGui.so+0xff
#71  0x7f46f052fbd8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib64/libQt5Core.so.5+0x128
#72  0x7f46f096b143 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib64/libQt5Gui.so.5+0x643
#73  0x7f46f094c8cc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt5Gui.so.5+0xac
#74  /usr/lib64/libQt5XcbQpa.so.5(+0x6b47e) [0x7f46da57b47e]
#75  /usr/lib64/libglib-2.0.so.0(g_main_context_dispatch+0x16f) [0x7f46ed2b3a5f]
#76  /usr/lib64/libglib-2.0.so.0(+0xa5a58) [0x7f46ed305a58]
#77  /usr/lib64/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7f46ed2b0e33]
#78  0x7f46f057c6f3 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt5Core.so.5+0x63
#79  0x7f46f052e57b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt5Core.so.5+0x12b
#80  0x7f46f05361b4 in QCoreApplication::exec() from /usr/lib64/libQt5Core.so.5+0x94
#81  0x7f46f511da19 in Gui::Application::runApplication() from /usr/local/lib64/libFreeCADGui.so+0x5df
#82  FreeCAD() [0x409147]
#83  /usr/lib64/libc.so.6(__libc_start_main+0xf2) [0x7f46efcb21e2]
#84  FreeCAD() [0x4084be]
```

Any ideas on why this would happen? When I save while not sketching, FreeCAD does not crash when saving.

Thanks!

/CN

christian-nils, 2021-02-15:
When compiling LinkStage3 at an earlier (random) commit: https://github.com/realthunder/FreeCAD/commit/a673d8a68801f4001121d7dda4002ddb7a6a45cb, this problem does not occur.

realthunder, 2021-02-15:
I can't reproduce the crash. Can you please send me your `link.user.cfg` file in `<your_home_dir>/.FreeCAD`.

christian-nils, 2021-02-16:
Hi @realthunder,

Thanks for your support, I do not have any `link.user.cfg` file. But I do have a [`user.cfg`](https://gist.github.com/christian-nils/e642498ac7a967db4a15b3c41df67b1f).

Is this useful for you?

Thanks!

christian-nils, 2021-02-16:
Hi again,

After checking different commits, it seems that the error starts with https://github.com/realthunder/FreeCAD/commit/bb3a7177a8c198703c933664d00b7240b665fcd5, I wonder if this is not related to my issue when compiling https://github.com/realthunder/FreeCAD/issues/92#issuecomment-778681290

Thanks for the help.

bgamari, 2021-02-16:
I believe this is the same issue that I am observing (noted initially in #462). The problem typically occurs during saving and can manifest in a few ways:

 * segmentation fault
 * a dialog with the text `std::bad_array:new_length`

It seems pretty clear that there is some memory unsafety afoot here. I have tried reproducing under `valgrind` but it seems that FreeCAD cannot run under `valgrind`. I will try to rebuild with ASAN/MSAN support in the hope that this turns up a hint.

realthunder, 2021-02-16:
> believe this is the same issue that I am observing (noted initially in [FreeCAD#462](https://github.com/FreeCAD/FreeCAD/pull/462)).

Base on the call stack, the crash happened when saving a thumbnail of your document. I did make some changes to rendering recently that might have caused this. But this looks different than realthunder/FreeCAD_Assembly3#462. Can you reproduce the crash on your side?

christian-nils, 2021-02-16:
Yes, that is correct, I do not think that the above-mentioned issue is related to https://github.com/realthunder/FreeCAD_assembly3/issues/462 since that was fixed with a recent commit.

@realthunder, do you see anything that I could do to help to fix this issue?

realthunder, 2021-02-16:
> @realthunder, do you see anything that I could do to help to fix this issue?

Can you reproduce the problem shown in above video? Make sure to enable thumbnail capturing with `Preference -> Document -> Save thumbnail into project file when saving document`

christian-nils, 2021-02-17:
Hi,
When `Preference -> Document -> Save thumbnail into project file when saving document` is disable, LinkStage3 does not crash. When it is enable, it does crash. I will just disable it for now :)
Thanks for the guidance.

ceremcem, 2021-02-17:
I can confirm that I *can't* reproduce the crash. FreeCAD does not crash on my side  while saving a document during sketch process, regardless of "Save thumbnail into project file" option's state. 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27228 (Git)
Build type: Release
Branch: LinkStage3
Hash: 99ccabfcf730689dfe1d027d90addd11332f193b
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

christian-nils, 2021-02-17:
Alright, thanks for checking. Below are my current setup.

```
OS: Fedora 33 (Workstation Edition) (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27228 (Git)
Build type: Release
Branch: LinkStage3
Hash: 99ccabfcf730689dfe1d027d90addd11332f193b
Python version: 3.9.1
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.0
Locale: English/United States (en_US)
```

christian-nils, 2021-02-18:
Hi,

This is fixed, it seemed to have been from the same source of error as https://github.com/realthunder/FreeCAD/issues/107.

Thanks @realthunder for your amazing work!

closed 2021-02-18 by christian-nils