realthunder/FreeCAD #106: SIGSEGV: Deleting a spinline on an image

byteofsoren, 2021-02-15:
Hello
Linkstage3 version of FreeCAD crashed when removing a polyline on the same plane as an image.
How to reproduce:
1. Start FreeCAD 
2. Create a new project.
3. Open Image wb. 
4. Create a planar image in 3D space, load any image file upon any plane with offset=0.
5. Open Part Design wb.
6. Create a new body.
7. Create a new sketch on the same plane.
8. Create a triangle with a spin line in the sketch.
9. delete one of the sides of the triangle.
10. Application has a 10% to crash; some times, it crashes some times it doesn't.

[image]

A possible workaround is to change the offset to -0.001 or put the image on another plane.
Thank you for your effort.

realthunder, 2021-02-15:
I can't reproduce the problem. Can you please attach a sample file here? Does the crash happen without the image? If you are using Linux, can you please provide the crash stack trace?

BTW, what is a spinline?

byteofsoren, 2021-02-15:
Ops. I meant polyline. 
A logfile from the crash:
[22:49:36.log](https://github.com/realthunder/FreeCAD/files/5984276/22.49.36.log)
Having a problem generating the trace file right now, I will lock into that tomorrow, probably something missing on my computer. 
The file I used:
[test1.zip](https://github.com/realthunder/FreeCAD/files/5984354/test1.zip)
Open the file and delete one of the polylines.
It won't work event time, and perhaps they need to be created, again and again, its about 5 to 10% chance that it crashes FreeCAD.
I have also tried with other images but can trigger the crash with even other file formats.
Thanks!

realthunder, 2021-02-16:
Thanks for the log. This problem has already been fixed in realthunder/FreeCAD#103. I'll make a new release some time this week.

luzpaz, 2022-01-18:
Close ticket

closed 2022-01-18 by realthunder