realthunder/FreeCAD #107: Segmentation fault when trying to do a dragging selection in Sketcher WB

christian-nils, 2021-02-17:
Hi,

LinkStage3 crashes when I try to select some sketch component using the dragged-rectangle selection. Here is the error stack:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib64/libc.so.6(+0x3da60) [0x7f1c13d05a60]
#1  0x7f1c15c89875 in SoAction::traverse(SoNode*) from /usr/local/lib64/libCoin.so.80+0x15
#2  0x7f1c196353a6 in Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) from /usr/local/lib64/libFreeCADGui.so+0x238
#3  0x7f1c15c8a481 in SoAction::apply(SoPathList const&, int) from /usr/local/lib64/libCoin.so.80+0x541
#4  0x7f1c19634ced in Gui::SoBoxSelectionRenderActionP::apply(Gui::SoBoxSelectionRenderAction*, SoPathList const&, int) from /usr/local/lib64/libFreeCADGui.so+0x195
#5  0x7f1c19634d41 in Gui::SoBoxSelectionRenderAction::apply(SoPathList const&, int) from /usr/local/lib64/libFreeCADGui.so+0x31
#6  0x7f1c15c90035 in SoGLRenderActionP::renderSingle(SoNode*) from /usr/local/lib64/libCoin.so.80+0x245
#7  0x7f1c15c904df in SoGLRenderActionP::render(SoNode*) from /usr/local/lib64/libCoin.so.80+0xff
#8  0x7f1c196353a6 in Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) from /usr/local/lib64/libFreeCADGui.so+0x238
#9  0x7f1c15c89d9a in SoAction::apply(SoNode*) from /usr/local/lib64/libCoin.so.80+0x2aa
#10  0x7f1c19634236 in Gui::SoBoxSelectionRenderAction::apply(SoNode*) from /usr/local/lib64/libFreeCADGui.so+0x32
#11  0x7f1c1972fab6 in Gui::View3DInventorViewer::renderToFramebuffer(QOpenGLFramebufferObject*) from /usr/local/lib64/libFreeCADGui.so+0x2b4
#12  0x7f1c1972f095 in Gui::View3DInventorViewer::grabFramebuffer() from /usr/local/lib64/libFreeCADGui.so+0x215
#13  0x7f1c1972ede0 in Gui::View3DInventorViewer::setRenderType(Gui::View3DInventorViewer::RenderType) from /usr/local/lib64/libFreeCADGui.so+0x2f6
#14  0x7f1b0ce9a425 in SketcherGui::ViewProviderSketch::mouseMove(SbVec2s const&, Gui::View3DInventorViewer*) from /usr/local/lib64/SketcherGui.so+0x184d
#15  0x7f1c1978d25e in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /usr/local/lib64/libFreeCADGui.so+0x4ae
#16  0x7f1c15e57edb in SoEventCallback::handleEvent(SoHandleEventAction*) from /usr/local/lib64/libCoin.so.80+0xab
#17  0x7f1c15c898b3 in SoAction::traverse(SoNode*) from /usr/local/lib64/libCoin.so.80+0x53
#18  0x7f1c15e1b9b8 in SoChildList::traverse(SoAction*, int, int) from /usr/local/lib64/libCoin.so.80+0x168
#19  0x7f1c15e5fb68 in SoGroup::doAction(SoAction*) from /usr/local/lib64/libCoin.so.80+0x48
#20  0x7f1c15e77ff8 in SoSeparator::doAction(SoAction*) from /usr/local/lib64/libCoin.so.80+0x28
#21  0x7f1c196060b1 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /usr/local/lib64/libFreeCADGui.so+0x53
#22  0x7f1c15c898b3 in SoAction::traverse(SoNode*) from /usr/local/lib64/libCoin.so.80+0x53
#23  0x7f1c15e1b9b8 in SoChildList::traverse(SoAction*, int, int) from /usr/local/lib64/libCoin.so.80+0x168
#24  0x7f1c15e5fb68 in SoGroup::doAction(SoAction*) from /usr/local/lib64/libCoin.so.80+0x48
#25  0x7f1c15e77ff8 in SoSeparator::doAction(SoAction*) from /usr/local/lib64/libCoin.so.80+0x28
#26  0x7f1c15c898b3 in SoAction::traverse(SoNode*) from /usr/local/lib64/libCoin.so.80+0x53
#27  0x7f1c15c92973 in SoHandleEventAction::beginTraversal(SoNode*) from /usr/local/lib64/libCoin.so.80+0xa3
#28  0x7f1c15c89d9a in SoAction::apply(SoNode*) from /usr/local/lib64/libCoin.so.80+0x2aa
#29  0x7f1c15e30e99 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /usr/local/lib64/libCoin.so.80+0x69
#30  0x7f1c196d34b6 in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /usr/local/lib64/libFreeCADGui.so+0x56
#31  0x7f1c196dbf39 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /usr/local/lib64/libFreeCADGui.so+0x195
#32  0x7f1c197311df in Gui::View3DInventorViewer::processSoEventBase(SoEvent const*) from /usr/local/lib64/libFreeCADGui.so+0x23
#33  0x7f1c196ef0e0 in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /usr/local/lib64/libFreeCADGui.so+0x28
#34  0x7f1c196f5853 in Gui::CADNavigationStyle::processSoEvent(SoEvent const*) from /usr/local/lib64/libFreeCADGui.so+0x1073
#35  0x7f1c196ef02b in Gui::NavigationStyle::processEvent(SoEvent const*) from /usr/local/lib64/libFreeCADGui.so+0x1db
#36  0x7f1c197311b5 in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /usr/local/lib64/libFreeCADGui.so+0x253
#37  0x7f1c196cc030 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /usr/local/lib64/libFreeCADGui.so+0x188
#38  0x7f1c1456d995 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib64/libQt5Core.so.5+0xa5
#39  0x7f1c15002eb2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x72
#40  0x7f1c15009eeb in QApplication::notify(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x4ab
#41  0x7f1c1929d55d in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/local/lib64/libFreeCADGui.so+0xff
#42  0x7f1c1456dbd8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib64/libQt5Core.so.5+0x128
#43  0x7f1c15008efa in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib64/libQt5Widgets.so.5+0x1da
#44  /usr/lib64/libQt5Widgets.so.5(+0x207375) [0x7f1c1505c375]
#45  /usr/lib64/libQt5Widgets.so.5(+0x20a6be) [0x7f1c1505f6be]
#46  0x7f1c15002ec3 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x83
#47  0x7f1c1929d55d in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/local/lib64/libFreeCADGui.so+0xff
#48  0x7f1c1456dbd8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib64/libQt5Core.so.5+0x128
#49  0x7f1c149a9143 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib64/libQt5Gui.so.5+0x643
#50  0x7f1c1498a8cc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt5Gui.so.5+0xac
#51  /usr/lib64/libQt5XcbQpa.so.5(+0x6b47e) [0x7f1bfe57747e]
#52  /usr/lib64/libglib-2.0.so.0(g_main_context_dispatch+0x16f) [0x7f1c112f1a9f]
#53  /usr/lib64/libglib-2.0.so.0(+0xa5a98) [0x7f1c11343a98]
#54  /usr/lib64/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7f1c112eee73]
#55  0x7f1c145ba6f3 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt5Core.so.5+0x63
#56  0x7f1c1456c57b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt5Core.so.5+0x12b
#57  0x7f1c145741b4 in QCoreApplication::exec() from /usr/lib64/libQt5Core.so.5+0x94
#58  0x7f1c1915ba09 in Gui::Application::runApplication() from /usr/local/lib64/libFreeCADGui.so+0x5df
#59  FreeCAD() [0x409147]
#60  /usr/lib64/libc.so.6(__libc_start_main+0xf2) [0x7f1c13cf01e2]
#61  FreeCAD() [0x4084be]
```

My setup is
```
OS: Fedora 33 (Workstation Edition) (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27228 (Git)
Build type: Release
Branch: LinkStage3
Hash: 99ccabfcf730689dfe1d027d90addd11332f193b
Python version: 3.9.1
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.0
Locale: English/United States (en_US)
```

Has anyone the same problem?

Thanks!

ceremcem, 2021-02-17:
If that's what you mean: 

[image]

I don't have any issues. My setup is: 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27228 (Git)
Build type: Release
Branch: LinkStage3
Hash: 99ccabfcf730689dfe1d027d90addd11332f193b
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

christian-nils, 2021-02-17:
Yes @ceremcem, Sorry I was very unclear! The click-n-drag selection makes my FreeCAD crashes. 
Since I have other issues at the moment I guess my setup is not properly done. Anyone else is running LinkStage with Qt.5.15.2 and Python3.9 ?

ceremcem, 2021-02-17:
I may setup a Debian Bullseye if you wish (I can't install Python 3.9 on Buster). Should we wait for realthunder's response first?

christian-nils, 2021-02-17:
That is very kind of you, thanks! But I think I have "solved" this issue. I have deleted .FreeCAD and started from scratch. It seems to be fine now.

Thanks again!

closed 2021-02-17 by christian-nils

realthunder, 2021-02-18:
The call stack indicates that this is the same issue you have experienced in #105. I can't imagine why deleting `.FreeCAD` can help in this case. Let's keep an eye on this and please report if there is any similar crashes.

BTW, since you are saying you don't have crash after deleting `.FreeCAD`. Maybe you can turn on thumbnail and see if it still crashes when saving while editing sketch.

christian-nils, 2021-02-18:
Yes, I did try to enable again the thumbnail saving, but it does continue to crash, unfortunately!

Thanks for the help!

realthunder, 2021-02-18:
Can you please try sync my repo and try again? Note sure if the above commit can fix the problem.

christian-nils, 2021-02-18:
Hi @realthunder!

Yes, it does fix it, see below the report view.
[image]
It shows the warning you have implemented in the last commit.