realthunder/FreeCAD #108: Workbench Path, in all versions, the problem of renaming operations

tiket18, 2021-02-19:
[image]
The essence of the problem is that I create an operation. In the menu for creating an operation there is no possibility to set the name of the operation. Therefore, first the operation is created, and then we give it a name.
But in the code, g-code, it may not be reflected. It may or may not be reflected. The regularity is not revealed. There was a time when it seemed that the situation would be corrected after rearranging the operations, but this hypothesis proved to be wrong. It seemed that "close-open" file corrects the situation, but it is not true.
Not critical, but annoying.
On the screenshot you can see that on the left in the tree of operations each operation has a name, and on the right, in the window of the operation code, the name is not changed.

By the way - it would be just great if the code view window would be indented instead of colored formatting. 
[image]

snelweg, 2021-03-08:
Isn't this more of a Path WB problem?
The developer of the Path WB is very responsive and fixes errors quick. 
Brad Collette @sliptonic

luzpaz, 2022-01-18:
@realthunder this ticket can be closed as it's not related to your branch.

sliptonic, 2022-01-19:
I've reviewed this on FreeCAD master.  The output is generated in the postprocessor file.  OP didn't mention which post he's using but at least for LinuxCNC it is outputting the object Label not the name.  So if he changes name in the tree, it should be reflected in the postprocessed gcode.  Also recommend closing this here.  
OP if you still have an issue, post to the FreeCAD Path/CAM section of the forum

luzpaz, 2022-01-19:
Thanks sliptonic. Closing ticked on your recommendation.

closed 2022-01-19 by luzpaz

tiket18, 2022-01-19:
The problem in the build of January 11 persists. I can't use another version because of Python10.

luzpaz, 2022-01-19:
>  OP if you still have an issue, post to the FreeCAD Path/CAM section of the forum  

@tiket18 :point_up: this is how you may get your Path issue addressed (note: sliptonic is the upstream Path WB main dev)