realthunder/FreeCAD #110: Crash on face selection

MuratUrsavas, 2021-03-02:
I'm using FreeCad LinkStage3 for couple of days and very pleased about it. Thanks for your great contribution to the project and community.

Today I was about to finish a design and tried to create a simple sketch on a planar face and just selected it. Then FreeCADLink crashed and disappeared.

Opened it again under GDB and then recreated the same scenario and crashed again. It seems problem is pretty reproducible. Since the file is company IP, I can not attach it publicly but can send it privately via email.

# Backtrace
```
A3 available
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7ffff5109210]
#1  0x7ffff4cc7f8d in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0x4bd
#2  0x7ffff75c65c5 in Gui::SoFCSelectionRoot::_renderPrivate(SoGLRenderAction*, bool, bool&) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/libFreeCADGui.so+0xb05
#3  0x7ffff75c68dd in Gui::SoFCSelectionRoot::renderPrivate(SoGLRenderAction*, bool) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/libFreeCADGui.so+0xcd
#4  0x7ffff4a9f1e9 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0x69
#5  0x7ffff4c5ff02 in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0x82
#6  0x7ffff4cccc21 in SoSwitch::doAction(SoAction*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0x2b1
#7  0x7ffff75c7ee6 in Gui::SoFCSwitch::doAction(SoAction*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/libFreeCADGui.so+0x596
#8  0x7ffff4cc7f9d in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0x4cd
#9  0x7ffff75c65c5 in Gui::SoFCSelectionRoot::_renderPrivate(SoGLRenderAction*, bool, bool&) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/libFreeCADGui.so+0xb05
#10  0x7ffff75c68dd in Gui::SoFCSelectionRoot::renderPrivate(SoGLRenderAction*, bool) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/libFreeCADGui.so+0xcd
#11  0x7ffff4cc7f9d in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0x4cd
#12  0x7ffff4cc7f9d in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0x4cd
#13  0x7ffff4a9f1e9 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0x69
#14  0x7ffff4aa6e36 in SoGLRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0x186
#15  0x7ffff75d3134 in Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/libFreeCADGui.so+0x64
#16  0x7ffff4a9fb62 in SoAction::apply(SoPathList const&, int) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0x5b2
#17  0x7ffff75d08ff in Gui::SoBoxSelectionRenderActionP::apply(Gui::SoBoxSelectionRenderAction*, SoPathList const&, int) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/libFreeCADGui.so+0x10f
#18  0x7ffff75d09f0 in Gui::SoBoxSelectionRenderAction::apply(SoPathList const&, int) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/libFreeCADGui.so+0xa0
#19  0x7ffff4aa6679 in SoGLRenderActionP::renderSingle(SoNode*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0x279
#20  0x7ffff4aa6b4e in SoGLRenderActionP::render(SoNode*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0x11e
#21  0x7ffff4aa6d52 in SoGLRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0xa2
#22  0x7ffff75d3134 in Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/libFreeCADGui.so+0x64
#23  0x7ffff4a9feca in SoAction::apply(SoNode*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0x2ea
#24  0x7ffff75d0cea in Gui::SoBoxSelectionRenderAction::apply(SoNode*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/libFreeCADGui.so+0x2a
#25  0x7ffff4d970fa in SoRenderManager::renderScene(SoGLRenderAction*, SoNode*, unsigned int) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0x4a
#26  0x7ffff4d981fc in SoRenderManager::actuallyRender(SoGLRenderAction*, int, int, int) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0x7c
#27  0x7ffff4d983b7 in SoRenderManager::renderSingle(SoGLRenderAction*, int, int, int) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0xc7
#28  0x7ffff4d989fb in SoRenderManager::render(SoGLRenderAction*, int, int, int) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0x9b
#29  0x7ffff4d9692d in SoRenderManager::render(int, int) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/../lib/libCoin.so.80+0x8d
#30  0x7ffff769ae62 in Gui::View3DInventorViewer::renderScene() from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/libFreeCADGui.so+0x262
#31  0x7ffff7653a75 in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/libFreeCADGui.so+0x325
#32  0x7ffff765a095 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/libFreeCADGui.so+0x45
#33  0x7ffff5e7f95d in QWidget::event(QEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Widgets.so.5+0x5a3
#34  0x7ffff5ee8db7 in QFrame::event(QEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Widgets.so.5+0x27
#35  0x7ffff6063ebe in QGraphicsView::viewportEvent(QEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Widgets.so.5+0x456
#36  0x7ffff7653b4c in SIM::Coin3D::Quarter::QuarterWidget::viewportEvent(QEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/libFreeCADGui.so+0x9c
#37  0x7ffff562c019 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Core.so.5+0xcd
#38  0x7ffff5e547e1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Widgets.so.5+0xcb
#39  0x7ffff5e5a31d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#40  0x7ffff730f399 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/libFreeCADGui.so+0x89
#41  0x7ffff562e606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Core.so.5+0x86
#42  0x7ffff5e7a612 in QWidgetPrivate::sendPaintEvent(QRegion const&) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Widgets.so.5+0x28
#43  /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Widgets.so.5(+0x17e91f) [0x7ffff5e5e91f]
#44  /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Widgets.so.5(+0x17efaf) [0x7ffff5e5efaf]
#45  0x7ffff5e6fdd4 in QWidgetPrivate::syncBackingStore() from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Widgets.so.5+0x66
#46  0x7ffff5e7fcd9 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Widgets.so.5+0x91f
#47  0x7ffff5f2a34f in QMainWindow::event(QEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Widgets.so.5+0xdd
#48  0x7ffff776a23c in Gui::MainWindow::event(QEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/libFreeCADGui.so+0x7c
#49  0x7ffff5e547f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#50  0x7ffff5e5a31d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#51  0x7ffff730f399 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib64/libFreeCADGui.so+0x89
#52  0x7ffff562e606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Core.so.5+0x86
#53  0x7ffff562e83e in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Core.so.5+0x1da
#54  /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/libQt5Core.so.5(+0x1dc54c) [0x7ffff565d54c]
#55  /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7ffff1cb598d]
#56  /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/./libglib-2.0.so.0(+0x54c21) [0x7ffff1cb5c21]
#57  /tmp/.mount_FreeCAKbYJyf/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7ffff1cb5cc1]
[... cut ...]
#63  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7ffff50ea0b3]
#64  /tmp/.mount_FreeCAKbYJyf/usr/bin/FreeCADLink(+0x47d9) [0x5555555587d9]
```

#Setup Information
```
OS: Ubuntu 20.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0226.23964 +3297 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 60580d5970c1033e0f962eb5317df80c049bf4a7
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)
```

realthunder, 2021-03-02:
A demo file will be extremely helpful. If you can't simplify it, please send it to me at realthunder.dev at gmail. Also please send me you user setting file at `~/.FreeCAD/link.user.cfg`.

MuratUrsavas, 2021-03-02:
Sent it, please check your inbox.

The related face is `Mirrored002.Face106`.

realthunder, 2021-03-02:
I have identified the problem. A proper fix will come soon. Meanwhile, you can disable `Show selection bounding box` option as a work around.

Your file also exposed another problem with VRML object. I am not sure if you have noticed, rendering your imported VRML PCB board with transparency is very slow. Looks like libCoin3D is not using OpenGL VBO for rendering VRML object with transparency. I'll work on that too.

MuratUrsavas, 2021-03-03:
Thank you for the fast response, I appreciate that.

> Your file also exposed another problem with VRML object. I am not sure if you have noticed, rendering your imported VRML PCB board with transparency is very slow.

You're right, I felt that, too. But thought that it could be due to a difference somewhere in the settings, as I knew LinkStage3 was using a different 3D engine.

MuratUrsavas, 2021-03-03:
By the way, it is especially slow if a sketch is being edited.

realthunder, 2021-03-03:
Yes, because EditAutoTransparency option is on, which forces other object to be transparent except the editing object. You can disable it in view options tool button.

[image]

realthunder, 2021-03-06:
This crash is fixed by commit dc87ffb2808023237a8f45f24f57ffdd3fada2d6. I messed up the commit message and thought it is reported in the asm3 repo.

MuratUrsavas, 2021-03-06:
No problem at all. Thanks for the prompt fix.

luzpaz, 2022-01-18:
@realthunder close this ticket?

closed 2022-01-18 by realthunder