realthunder/FreeCAD #111: SIGSEGV while drawing a sketch

MuratUrsavas, 2021-03-02:
While I was working on the project mentioned in #110 FreeCAD Link crashed with SIGSEGV.

Reproduction steps would be irrelevant because I just selected a face and started to draw a simple sketch. So it is related with file contents. As I mentioned in the other issue, I can send the file via email privately.

# Backtrace
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.FreeCADError'>: std::bad_array_new_length
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7ffff5109210]
#1  0x7ffff4cc7f8d in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/../lib/libCoin.so.80+0x4bd
#2  0x7ffff4a9f1e9 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/../lib/libCoin.so.80+0x69
#3  0x7ffff4aa6e36 in SoGLRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/../lib/libCoin.so.80+0x186
#4  0x7ffff75d3134 in Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/libFreeCADGui.so+0x64
#5  0x7ffff4a9fb62 in SoAction::apply(SoPathList const&, int) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/../lib/libCoin.so.80+0x5b2
#6  0x7ffff75d08ff in Gui::SoBoxSelectionRenderActionP::apply(Gui::SoBoxSelectionRenderAction*, SoPathList const&, int) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/libFreeCADGui.so+0x10f
#7  0x7ffff75d0979 in Gui::SoBoxSelectionRenderAction::apply(SoPathList const&, int) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/libFreeCADGui.so+0x29
#8  0x7ffff4aa6679 in SoGLRenderActionP::renderSingle(SoNode*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/../lib/libCoin.so.80+0x279
#9  0x7ffff4aa6b4e in SoGLRenderActionP::render(SoNode*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/../lib/libCoin.so.80+0x11e
#10  0x7ffff4aa6d52 in SoGLRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/../lib/libCoin.so.80+0xa2
#11  0x7ffff75d3134 in Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/libFreeCADGui.so+0x64
#12  0x7ffff4a9feca in SoAction::apply(SoNode*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/../lib/libCoin.so.80+0x2ea
#13  0x7ffff75d0cea in Gui::SoBoxSelectionRenderAction::apply(SoNode*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/libFreeCADGui.so+0x2a
#14  0x7ffff4d970fa in SoRenderManager::renderScene(SoGLRenderAction*, SoNode*, unsigned int) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/../lib/libCoin.so.80+0x4a
#15  0x7ffff4d981fc in SoRenderManager::actuallyRender(SoGLRenderAction*, int, int, int) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/../lib/libCoin.so.80+0x7c
#16  0x7ffff4d983b7 in SoRenderManager::renderSingle(SoGLRenderAction*, int, int, int) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/../lib/libCoin.so.80+0xc7
#17  0x7ffff4d989fb in SoRenderManager::render(SoGLRenderAction*, int, int, int) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/../lib/libCoin.so.80+0x9b
#18  0x7ffff4d9692d in SoRenderManager::render(int, int) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/../lib/libCoin.so.80+0x8d
#19  0x7ffff769ae62 in Gui::View3DInventorViewer::renderScene() from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/libFreeCADGui.so+0x262
#20  0x7ffff7653a75 in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/libFreeCADGui.so+0x325
#21  0x7ffff765a095 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/libFreeCADGui.so+0x45
#22  0x7ffff5e7f95d in QWidget::event(QEvent*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0x5a3
#23  0x7ffff5ee8db7 in QFrame::event(QEvent*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0x27
#24  0x7ffff6063ebe in QGraphicsView::viewportEvent(QEvent*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0x456
#25  0x7ffff7653b4c in SIM::Coin3D::Quarter::QuarterWidget::viewportEvent(QEvent*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/libFreeCADGui.so+0x9c
#26  0x7ffff562c019 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Core.so.5+0xcd
#27  0x7ffff5e547e1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0xcb
#28  0x7ffff5e5a31d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#29  0x7ffff730f399 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib64/libFreeCADGui.so+0x89
#30  0x7ffff562e606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Core.so.5+0x86
#31  0x7ffff5e7a612 in QWidgetPrivate::sendPaintEvent(QRegion const&) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0x28
#32  0x7ffff5e7ad3f in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0x707
#33  0x7ffff5e7b4b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#34  0x7ffff5e7b1be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#35  0x7ffff5e7b4b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#36  0x7ffff5e7b1be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#37  0x7ffff5e7b4b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#38  0x7ffff5e7b1be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#39  0x7ffff5e7b4b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#40  0x7ffff5e7b1be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#41  0x7ffff5e7b4b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#42  0x7ffff5e7b1be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#43  0x7ffff5e7b4b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#44  0x7ffff5e7b1be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#45  0x7ffff5e7b4b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#46  0x7ffff5e7b3ce in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA1lBtxd/usr/bin/../lib/libQt5Widgets.so.5+0x1de
[... cut ...]
#82  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7ffff50ea0b3]
#83  /tmp/.mount_FreeCA1lBtxd/usr/bin/FreeCADLink(+0x47d9) [0x5555555587d9]
```
# Setup Information
```
OS: Ubuntu 20.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0226.23964 +3297 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 60580d5970c1033e0f962eb5317df80c049bf4a7
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)
```

ceremcem, 2021-03-02:
There are a few commits which might be relevant since `60580d5970`. Some of us are using FreeCAD directly by compiling from source (with slightly different environments), so if you could reproduce the problem with a non IP version of the project file, we might help in some way by trying to reproduce the problem in our side, just like https://github.com/realthunder/FreeCAD_assembly3/issues/472

One option might be removing objects from your original project until you could still reproduce the problem. This stripped down version might not be considered IP.

MuratUrsavas, 2021-03-02:
Unfortunately simplification prevented the reproduction of the issuI've sent the file to @realthunder and I can send it to you, too, if you want.

By the way the target face is Chamfer0004.Face15.

ceremcem, 2021-03-02:
I think realthunder is on the issue now. Rarely issues can not be reproduced on his side, maybe sending the project file to me makes sense at that point.

luzpaz, 2022-01-18:
@MuratUrsavas can you reproduce?