realthunder/FreeCAD #114: AppImage doesnot update to latest build

anirudhnehra, 2021-03-08:
Hello,
Great job alaround.
You have said that AppImage can automatically update to latest daily build through preferences>appimage>update but it does not work. 
Even after running the update I get the following data:
OS: Debian GNU/Linux 10 (buster) (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.11.16.22739 +2766 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 6730ad3b3bd841c2745ec98bf953c523dea86913
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/India (en_IN)

[image]

snelweg, 2021-03-08:
Same here, it never worked for me.
When I choose the AppImage download from preferences the file gets downloaded and is placed in the Application folder. But it is not integrated into AppImage. The menus in Debian are not adjusted too. So I end up with loads of AppImages floating around in my Applications.
[image]
When I start the newest version and choose integration into AppImage then it will be started and I have another menu entry. 
[image]

So I just download the releases by hand and integrate them myself and delete the old images that I do no longer use. That takes care of the inflated menu entries too.
I never found the possibility to update with AppImage from within FreeCAD a worthwhile one. It never worked for me and so I do not bother with it anymore. I would use it if it worked, but as it is now I will not.

> OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0308.24301 +3361 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: d4e0058e12c27326da1a27501356fad9cb9f604b
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

btw: Thanks for the new splash screen, looks really fresh

luzpaz, 2022-01-18:
We never got the upstream (FreeCAD master) in-app appimage updater to work properly either. So it's recommended to either use the GUI or the CLI methods of [auto-updating](https://wiki.freecad.org/AppImage#Automatic_updating). There is a new method that's going to be available soon using Snap packages (See https://github.com/realthunder/FreeCAD/pull/254). It will probably be working on the next Link release and will be announced then. 

Closing this ticket in the meantime.

closed 2022-01-18 by luzpaz