realthunder/FreeCAD #116: Dual  monitor, preferences opens centered between monitors

snelweg, 2021-03-08:
A cosmetic issue, but sometime quite annoying nonetheless.

When on a dual monitor setup, FC on its own monitor, then opening the preferences dialog opens exactly in the middel between the monitors.
Would be nicer if it did open on the active FC monitor.
(the gray part in the video is the left monitor, the right monityor is where FC runs)

https://user-images.githubusercontent.com/1257901/110365235-134f6080-8045-11eb-87c3-8119c057650b.mp4

dickshaydle, 2021-03-17:
I also find this pretty annoying. Every time I open preferences I have to move it because it is half way outside the visible area because it is displayed on the smaller monitor.

What about saving the last window position?

realthunder, 2021-03-18:
Can you guys post more details? Your FreeCAD About information. Screen size of both of your monitors. Which one is the primary screen. I can't reproduce the problem. The preference window position is remembered and restored on open. And it always pop up on my primary screen regardless of where FreeCAD main window is.

snelweg, 2021-03-18:
oops
Main (primary) monitor (the one on the right) 1920x1080, left one 1360x768
On my setup the Preferences screen always opens up right between the two monitors, as if it centers on the total width of both monitors. I am using XFCE as window manager, maybe that causes problems? 
Even after restarting FC the same positioning.

> OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.311.24301 +3373 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 91ca94db328bf6126c4a01547b6aa2202d876ebc
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

wohltat, 2021-03-18:
[image]

The black part on the top left is outside of the visible screen. The left screen is the primary monitor.

The preference window is always opened to the same position, no matter what i do. So far, i have not seen any modal window of which the position or size is saved/restored.

```
OS: Linux Mint 20.1 (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.311.24301 +3373 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 91ca94db328bf6126c4a01547b6aa2202d876ebc
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)
```

snelweg, 2021-03-18:
> The preference window is always opened to the same position, no matter what i do. So far, i have not seen any modal window of which the position or size is saved/restored.

But your screen looks different then mine. The dialog box opens on your primary  monitor centered horizontally. On my setup it opens centered between monitors. Maybe there is indeed same saved setting somewhere that 'remembers' old values.

realthunder, 2021-03-18:
That's strange. Do you still keep the old .FreeCAD directory? If so please post here your link.user.cfg file inside.

snelweg, 2021-03-18:
> That's strange. Do you still keep the old .FreeCAD directory? If so please post here your link.user.cfg file inside.

Yes I kept it. Switching back to the saved .FreeCAD directory same problem again.

`<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<FCParameters>
  <FCParamGroup Name="Root">
    <FCParamGroup Name="BaseApp">
      <FCParamGroup Name="Preferences">
        <FCParamGroup Name="General">
          <FCBool Name="SaveUserParameter" Value="1"/>
          <FCText Name="LastModule">PartDesignWorkbench</FCText>
          <FCText Name="FileOpenSavePath">/home/xx/Nextcloud/FreeCAD/Jesca</FCText>
          <FCText Name="AutoloadModule">PartDesignWorkbench</FCText>
          <FCBool Name="ShowSplasher" Value="1"/>
          <FCBool Name="PythonWordWrap" Value="1"/>
          <FCInt Name="ToolbarIconSize" Value="24"/>
          <FCText Name="FileImportFilter">All files (*.*)</FCText>
          <FCBool Name="ConfirmAll" Value="1"/>
          <FCText Name="OffscreenImageFormat">png</FCText>
          <FCInt Name="OffscreenImageBackground" Value="3"/>
          <FCText Name="FileExportFilter">STEP with colors (*.step *.stp)</FCText>
          <FCText Name="PreferenceGeometry">957 0 960 926</FCText>
          <FCBool Name="RecentIncludesImported" Value="1"/>
        </FCParamGroup>
        <FCParamGroup Name="Macro">
          <FCBool Name="LocalEnvironment" Value="1"/>
          <FCText Name="MacroPath">/home/xx/.FreeCAD/Macro</FCText>
          <FCBool Name="RecordGui" Value="1"/>
          <FCBool Name="GuiAsComment" Value="1"/>
          <FCBool Name="ScriptToPyConsole" Value="1"/>
          <FCBool Name="ScriptToFile" Value="0"/>
          <FCText Name="ScriptFile">FullScript.FCScript</FCText>
        </FCParamGroup>
        <FCParamGroup Name="Mod">
          <FCParamGroup Name="Import">
            <FCParamGroup Name="hSTEP">
              <FCBool Name="ReadShapeCompoundMode" Value="0"/>
            </FCParamGroup>
            <FCBool Name="UseAppPart" Value="1"/>
            <FCBool Name="UseLegacyImporter" Value="0"/>
            <FCBool Name="UseBaseName" Value="1"/>
            <FCBool Name="ImportHiddenObject" Value="1"/>
            <FCBool Name="ExportHiddenObject" Value="1"/>
            <FCBool Name="ExportKeepPlacement" Value="1"/>
            <FCBool Name="ReduceObjects" Value="0"/>
            <FCBool Name="ShowProgress" Value="1"/>
            <FCBool Name="ExpandCompound" Value="1"/>
            <FCBool Name="ExportLegacy" Value="0"/>
            <FCInt Name="ImportMode" Value="0"/>
          </FCParamGroup>
          <FCParamGroup Name="OpenSCAD"/>
          <FCParamGroup Name="Part">
            <FCParamGroup Name="Boolean">
              <FCBool Name="CheckModel" Value="0"/>
              <FCBool Name="RefineModel" Value="0"/>
            </FCParamGroup>
            <FCParamGroup Name="IGES">
              <FCInt Name="Unit" Value="0"/>
              <FCBool Name="BrepMode" Value="0"/>
              <FCBool Name="SkipBlankEntities" Value="1"/>
              <FCText Name="Company"/>
              <FCText Name="Author"/>
            </FCParamGroup>
            <FCParamGroup Name="General">
              <FCInt Name="WriteSurfaceCurveMode" Value="1"/>
            </FCParamGroup>
            <FCParamGroup Name="STEP">
              <FCInt Name="Unit" Value="0"/>
              <FCText Name="Scheme">AP214IS</FCText>
[... 1572 lines cut ...]
            <FCText Name="Top">Break,View,Assembly3 Constraints,Assembly3 Constraints2,Assembly3 Sketch Constraints,Assembly3 Measurements,Assembly3 Navigation,Break,Assembly3</FCText>
            <FCText Name="Left">Tabs</FCText>
            <FCText Name="Right"/>
            <FCText Name="Bottom"/>
          </FCParamGroup>
          <FCParamGroup Name="ImageWorkbench"/>
          <FCParamGroup Name="PathWorkbench"/>
          <FCParamGroup Name="SpreadsheetWorkbench"/>
          <FCParamGroup Name="SMWorkbench"/>
          <FCParamGroup Name="CADENAS3DfinditWorkbench"/>
        </FCParamGroup>
        <FCParamGroup Name="System">
          <FCParamGroup Name="CADENAS3DfinditWorkbench"/>
        </FCParamGroup>
      </FCParamGroup>
    </FCParamGroup>
    <FCParamGroup Name="Plugins">
      <FCParamGroup Name="addonsRepository">
        <FCBool Name="readWarning" Value="1"/>
      </FCParamGroup>
    </FCParamGroup>
  </FCParamGroup>

</FCParameters>
`

wohltat, 2021-03-18:
There is actually a place where the position and size is stored.
_link.user.cfg:_
`<FCText Name="PreferenceGeometry">683 0 683 708</FCText>`

It is saved correctly but on restoring there is something going wrong:
- The window is pulled to the first monitor even though i placed it on the second monitor before. 
- y position seems to be always set to "0" (which causes it to be outside visible area on my setup)
- The x/y size, x-position is restored correctly if the window stays on the first monitor
- if placed on the second monitor, the x/y size is reduced so it fits on the first monitor

realthunder, 2021-03-18:
The y position is not restored on Linux, because of some [peculiarity](https://doc.qt.io/qt-5/application-windows.html#x11-peculiarities) in Linux desktop environment. Can you try move the preference dialog a little bit, and then close it. It should save a non-zero value for y. Nevertheless, the saved y value is ignored during restore.

So the problem is mainly that Qt always show pop up window on the primary monitor, and if it has a lower resolution than the secondary monitor and the FreeCAD main window is in the secondary monitor, things get a little weird.

wohltat, 2021-03-18:
> So the problem is mainly that Qt always show pop up window on the primary monitor, and if it has a lower resolution than the secondary monitor and the FreeCAD main window is in the secondary monitor, things get a little weird.

Looks **line 439** is why only one monitor is considered. I would guess that "desktop()" uses the entire virtual desktop, that you can see in the screenshot of my previous post. This doesn't seem like a general Qt limitation to me, since the window is moved in the virtual desktop coordinates (window is outside the first monitor). 
https://github.com/realthunder/FreeCAD/blob/acc8717c2b634e573ee8dfdc896372e970a2a310/src/Gui/DlgPreferencesImp.cpp#L438-L442

-------------

> The y position is not restored on Linux, because of some peculiarity in Linux desktop environment. Can you try move the preference dialog a little bit, and then close it. It should save a non-zero value for y. Nevertheless, the saved y value is ignored during restore.

As i said previously. The saving of the values works flawlessly. It's just the restoring of the values.
I can't see what the linked "peculiarities" have to do with this. As i read it, it only talks about why the size of the window is not predictable. Furthermore, why should it be different then the x-axis?
I can't understand what __line 477__ is good for. I'm pretty sure it is causing the problem i've seen.

https://github.com/realthunder/FreeCAD/blob/acc8717c2b634e573ee8dfdc896372e970a2a310/src/Gui/DlgPreferencesImp.cpp#L472-L479

realthunder, 2021-03-18:
`desktop()` code is only used in Qt4. Qt5 code uses the primary screen geometry, and it is only used to limited preference widget position. In all cases, the widget shouldn't be moved outside of the desktop.

The problem with X window is about the window decoration, i.e. the window border, title bar and stuff. Qt says it has trouble finding out the size of them. The net effect, at least when I tried on Ubuntu Gnome is that if I restore the Y value, the preference dialog will noticeably moving downward a bit each time you open it. So I decided to only restore X value and use the current Y. But then, X Window is asynchronous, by the time the move command reaches the server, something might have already happened. And that something may also be the reason why your dialog moved outside of your screen. I guess using a timer to delay the restore may be a solution. Not sure though.

wohltat, 2021-03-19:
>  In all cases, the widget shouldn't be moved outside of the desktop. ...

I agree. But it obviously is moved outside. I think using only the primary monitor geometry is not sufficient. Using `desktop()` should still work but since it is deprecated, some other solution would be preferable. For example maybe the following
https://doc.qt.io/qt-5/qscreen.html#availableVirtualGeometry-prop

> The problem with X window is about the window decoration ...

To get the displacement offset, one could try to perform a "save-restore-cycle". This means set coordinates, get coordinates and store the difference.

wohltat, 2021-03-20:
Much better now. But there are still some problems.
On the first monitor the window is squished in y-direction and on the second in x-direction.
Furthermore the window is force to be on the same monitor as the main window.

Personally i would prefer to simply be able to put it anywhere and it being restored that way without repositioning/resizing.
Just removing the constraints does the trick, since Qt somehow already cares for the window not being outside of the screen.

```c++
void DlgPreferencesImp::restoreGeometry()
{
    if (geometryRestored)
        return;
    geometryRestored = true;

    std::string geometry = App::GetApplication().GetParameterGroupByPath(
            "User parameter:BaseApp/Preferences/General")->GetASCII(
                "PreferenceGeometry", "");
    std::istringstream iss(geometry);
    int x,y,w,h;
    if (iss >> x >> y >> w >> h) {
        this->move(x, y);
        savedPos = QPoint(x, y);
        QMetaObject::invokeMethod(this, "resizeWindow",
                Qt::QueuedConnection,
                QGenericReturnArgument(),
                Q_ARG(int, w),
                Q_ARG(int, h));
    }
}
```

I tested it and it works fine on my system (Ubuntu 20.04, XFCE). But i don't how know this behaves on windows.

wohltat, 2021-03-23:
Just tested it and it works great in every possible constellation of the windows.

wohltat, 2021-08-23:
@snelweg i guess this issue can be closed

closed 2021-08-23 by snelweg