realthunder/FreeCAD #119: Sketching point selection problematic when drawing on bottom of plane

jpedrick, 2021-03-28:
Reproduce by starting a drawing in the XY plane, flip the plane around so you're drawing from the -Z direction and try to draw and select some points.

Resolved by going back to +z perspective.

Using linkstage3 2021-03-26 Appimage on Ubuntu 20.10.

realthunder, 2021-03-28:
The reason for the difficulty to select is because that each geometry element has a predefined z order. The mouse click will only pick the nearest one. You can pick overlapped geometry using the `G, G` shortcut. But I am not sure why you want to draw with plane flipped. May I ask why you want to do that?

jpedrick, 2021-03-30:
@realthunder seems more complicated than just the z order, as I was unable to select any points when the z-was flipped. My expectation was that the selection would use the closes point to the viewpoint. It's been some time since I did selection using OpenGL, but I recall OpenGL (maybe using glRenderMode(GL_SELECT) ) has support for this built in.

I stopped trying to draw from the -z perspective, but felt like a bug that might frustrate users so I reported it. I had created my sketch on a face that later required flipping all the dimensions/features, probably because I'm a FreeCAD novice. :D

jpedrick, 2021-04-05:
@realthunder I'm realizing the reason I was drawing on the back of the plane was due the way the latest linkstage 3 build orients the drawing plane. It's somehow not always in the + normal direction to the attachment plane. For example, you select a face of a cube to attach a sketch to and it places the body of the cube in front so you're drawing "inside" the cube instead of on the outside of the face. I have been rotating the model around so I can draw from the outside perspective instead of inside, which causes the selection issue to be a problem. I'm not sure how to consistently reproduce the issue.