realthunder/FreeCAD #120: Preferences - Display - Draw styles tab crashes the program in the second monitor

bitacovir, 2021-03-30:
I use a second monitor (DELL 23,8") connected to my laptop. Both screens working as Extended displays with 1920x1080.
When I have FC displayed in the 2nd screen (Dell monitor) it crashes if I click on Preferences - Display - Draw styles tab.
But, if FC is opened in the laptop' screen, click on Draw styles tab does not produce any crash in the program.

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.328.24301 +3441 (Git)
Build type: Release
Branch: LinkStage3
Hash: 0a157a307c91d241a370cdda6d8f0471b7c48915
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.5.0
Locale: English/United States (en_US)

luzpaz, 2022-01-18:
@bitacovir dual monitor support has been improved in latest Daily. Can you test and report back?

bitacovir, 2022-01-18:
@luzpaz 
Hi. Now it is working normally in the second monitor. Great work!!
You can close this comment.

luzpaz, 2022-01-18:
@bitacovir i can't close it since I don't have permissions. Please close it yourself since you are OP :wink:

closed 2022-01-18 by realthunder