realthunder/FreeCAD #121: Shape2DView does not recompute when Arch Section is from a BuildingPart

bitacovir, 2021-03-30:
Shape2DView does not recompute when Arch Section is from a BuildingPart

In FC 0.19 create a BuildingPart containing two walls.
Create an Arch Section from the Building Part
Create a Shape2DView from the Arch section.
Save and open the file with FC LinkStage3
Recompute fails with Shape2DView element

[image]
 Error in Report view panel: 

`18:59:18  <TopoShape> shape2dview.py(128)|TopoShapeEx.cpp(1003): Null input shape
18:59:18  <TopoShape> shape2dview.py(128)|TopoShapeEx.cpp(1003): Null input shape
18:59:18  <TopoShape> shape2dview.py(128)|TopoShapeEx.cpp(1003): Null input shape
18:59:18  <TopoShape> shape2dview.py(128)|TopoShapeEx.cpp(1003): Null input shape
18:59:18  <TopoShape> shape2dview.py(128)|TopoShapeEx.cpp(1003): Null input shape
18:59:18  <Exception> shape2dview.py(128)|TopoShapeEx.cpp(1010): Null shape
18:59:18  Traceback (most recent call last):
  File "C:\Users\rafmo\Documents\Programs2019\FreeCAD-asm3-Win64-Py3-Qt5.12-20210328\Mod\Draft\draftobjects\shape2dview.py", line 223, in execute
    obj.Shape = self.getProjected(obj,comp,proj)
  File "C:\Users\rafmo\Documents\Programs2019\FreeCAD-asm3-Win64-Py3-Qt5.12-20210328\Mod\Draft\draftobjects\shape2dview.py", line 128, in getProjected
    return Part.makeCompound(edges)
<class 'RuntimeError'>: FreeCAD exception thrown (Null shape)
18:59:18  Recompute failed! Please check report view.`

realthunder, 2021-03-31:
Probably related to realthunder/FreeCAD_Assembly3#538. Can you please provide a file showing the problem? When the recomputation fails, can you please manually mark the whole document for recompute (right click document icon in tree view, and select 'Mark recompute'), and recompute again?

bitacovir, 2021-03-31:
Even if you "Mark to recompute" the document, it fails.
Here an example.

[Shape2Dview.zip](https://github.com/realthunder/FreeCAD/files/6235845/Shape2Dview.zip)