realthunder/FreeCAD #122: Circular Loft Results in Invalid Shape

SamuelFarinas, 2021-04-03:
Hi, I created a circular loft with detail under 3 mm and when running Part->Check Geometry in the Part Workbench, I get a message that the Edge2 has an error message "check failed". It seems to be a regression because I did not encounter this geometry check error in the main FreeCAD branch. Is there any setting I can use to avoid this error? I have attached a simple model with this weird error.
[Broken Loft.zip](https://github.com/realthunder/FreeCAD/files/6251948/Broken.Loft.zip)

realthunder, 2021-04-04:
Hmm... I get the same error using upstream FreeCAD. Note that you'll have to reset the second sketch attachment when you open the file in upstream. You can mark the whole document for recomputing to refresh the shape.

I am not exactly sure why it is considered as an error. It seems that if you build loft with small scale, such as a few mm, you'll get this error. If it is bigger, like > 10mm, then no error. The relevant OCC error code is `CheckPolygonOnTriangulation`. The relevant function that report this error is [here](https://dev.opencascade.org/doc/refman/html/class_b_rep_check___edge.html#a34237e7b8a6eed82a4f87c828e19d7b5). 

Maybe you can raise the issue in the FreeCAD forum.

SamuelFarinas, 2021-04-04:
My apologies, my upstream version was 0.19 (2020/12/27). I reset, recomputed, ran the check again on 0.20 (2021/03/28) and confirmed the check fails on that one, too. I'll report this issue on the FreeCAD forum. Thanks!

closed 2022-01-18 by luzpaz