realthunder/FreeCAD #124: Two interface problems in the Path workbench

tiket18, 2021-04-08:
Appears in all assemblies.

1. The workbench does not react to changes in part placement
2.  All the time an unnecessary (for work) item - workpiece - is shown visible.

Explanation.

1. Take 2-10 any bodies (for example, square, circle, etc.). In the machine we will make them from one workpiece. To do this, move these bodies (link) as they will lie on the machine table. After that, create a union of these bodies (it is necessary to do so in order to be able to process the part from both sides). Let's create a new job with this body (job) = this will automatically create the Model. So, if we change the square (circle, etc.) - its dimensions, thickness, etc., these changes will be displayed in the model. But if we change the position of the square (circle, etc.) in the overall layout, the Model will not display it. We have to delete the job and create a new one.

2. Here we have a Model. Select an edge, create a path = open the create menu. Ignoring previous visibility settings, I get in the way of newly created Stock. I have to switch to the Tree tab, disable visibility (display) of Stock. Very interferes = stok interferes with selection of required elements for processing when Path job edit menu is open

luzpaz, 2022-01-18:
Missing [About](https://wiki.freecad.org/About) info.  
Also not sure what this is the issue being described in this ticket.

luzpaz, 2022-04-11:
@tiket18 please respond

tiket18, 2022-04-12:
В текущем году с январского слияния с основной веткой верстак Path в данной сборке вообще не релевантен. Я просил сделать слияние с основной веткой, но был проигнорирован, что для разработчика данной сборки необычно - предположу лишь, что это игнорирование в связи с моими Российскими корнями.
Сейчас я использую верстак Path из appimage основной ветки, и по состоянию на сегодня этот Path работает лучше всех виденных мною других его версий (там в этом году было очень много полезных изменений).

This year, since the January merge with the main branch, the Path workbench in this build is not relevant at all. I asked to merge with the main branch but was ignored, which is unusual for a developer of this build - I'll just assume this ignoring is due to my Russian roots.
I am using Path workbench from main branch appimage now, and as of today this Path works better than any other version I've seen (there were many useful changes this year).

Translated with www.DeepL.com/Translator (free version)

closed 2022-04-12 by tiket18

luzpaz, 2022-04-12:
@tiket18 
...or perhaps it's a less paranoid reason...? Maybe it's because just one person is programming/maintaining an experimental branch/fork/copy of FreeCAD and has no time to maintain and keep all workbenches up to date? (that means more bugs due to incompatibilities with  LinkStage/LinkDaily besides the bugs the have already)

How realistic is that?  
https://wiki.freecadweb.org/Workbenches/ru
https://wiki.freecad.org/External_workbenches/ru

tiket18, 2022-04-12:
> ...or perhaps it's a less paranoid reason...?

How without it, without paranoia in the circumstances of general bullying. But perhaps you are right - I would be glad to be wrong, and so I will paraphrase, see above.

Nevertheless, the question of regularity of mergers is acute - the Master alone does a great job, but can not embrace the immensity, and it is a lot of other modules. The Path workbench is what eventually turns an idea into a real tangible object, without which everything made (by the program user) = just "fun pictures".

luzpaz, 2022-04-12:
I see and I understand your frustration. The point here is that Link branch isn't a traditional fork of FreeCAD where it's a separate project all together. realthunder has signaled that he doesn't intend for it to be that. More precisely, my understanding is it's a way for him to implement his ideas at his own pace and his own constraints (there is more shall we say 'bureaucracy' in the upstream branch but it's more complex than that etc...). He is testing and improving Topologicial naming, experimental renderers, UI/UX macros and features, so so much.

tiket18, 2022-04-12:
That's exactly! - that this approach requires regular merges with the main branch. Until the new year, this was happening quite regularly. It's now April, and the last merge was, if memory serves me correctly, in early January.

luzpaz, 2022-04-12:
Well, glad we're on the 'same page'