realthunder/FreeCAD #125: "Select a File" window freezes FreeCAD

bitacovir, 2021-04-11:
If I open a Select a File window, this freezes FreeCAD (Not Responding window).

Case 1:
In DraftWB Create a ShapeString object and select it.
Go to Property - Data- Font File, 
Click on "..." 
The "Select a File" window will freeze. 

Case 2:
Create a TechDraw page.
Select the template object
Go to Property - Data- Template, 
Click on "..." 
The "Select a File" window will freeze. 

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.328.24301 +3441 (Git)
Build type: Release
Branch: LinkStage3
Hash: 0a157a307c91d241a370cdda6d8f0471b7c48915
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.5.0
Locale: English/United States (en_US)

[image]

luzpaz, 2022-01-18:
@bitacovir can you still reproduce?

bitacovir, 2022-01-18:
@luzpaz 
Hi. Nop. Everything is working normally now...

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)

luzpaz, 2022-01-18:
Thanks, pls close ticket :wink:

closed 2022-01-18 by realthunder