realthunder/FreeCAD #130: Step file origin gone after importing

hdt551, 2021-04-22:
Importing a step file with:
Version: 2021.412.22506 +5327 (Git) AppImage
Build type: Release
Branch: LinkDaily

[image]

Importing same step file with:
Version 0.20, Revision number 24720

[image]

Lack of origin causes errors if you move from your LinkDaily back to the official version.

realthunder, 2021-04-26:
I'll add an option to force create origin in App::Part. Meanwhile you can use the following python code (using my branch) to do just that (explanation: the origin object will auto create the missing origins if there is any code accessing them).

```python
for obj in App.ActiveDocument.Objects:
  if obj.isDerivedFrom('App::Origin'):
    _ = obj.OriginFeatures
```

realthunder, 2021-04-26:
Oh, forgot about that. There is already a parameter controlling this behavior. Boolean parameter named `CreateOrigin` in path `BaseApp/Preference/Group`. Create this parameter if it's not there and set it to `True`.

hdt551, 2021-04-26:
I added CreateOrigin and it worked fine.

Thanks

luzpaz, 2022-07-04:
@realthunder good to close, right?

closed 2022-07-04 by luzpaz