realthunder/FreeCAD #133: body get hidden when import model to A2+ WB

QuakeString, 2021-04-30:
Cant see some part of body in A2+ work bench. I'm not sure if I'm trying something wrong. or its issue of A2+ Workbench and may be I need to report to A2+ Workbench github. Your advice will be appriciated. [Here is link of FreeCAD model file.](https://www.dropbox.com/sh/443isukcqjyixh9/AADvBI94SMiH7LX1imSGIHkla?dl=0)

[image]
[image]

realthunder, 2021-05-02:
I have submit a [PR](https://github.com/kbwbe/A2plus/pull/428) to A2plus for the fix.

kbwbe, 2021-05-02:
The PR has been merged to A2plus...

QuakeString, 2021-05-09:
Could it be the issue for the same A2+ WB or something else ?
FreeCAD Using too much memory and its also performing poorly. 
[image]
I've opened the same model with FreeCAD 0.19.1, using less then half memory and its also very fast to use, there is no lag in zooming.
[image]

realthunder, 2021-05-09:
Can you please paste the FreeCAD about information here? Are you using my Stable release? If so, can you please try Daily?

QuakeString, 2021-05-09:
OS: Manjaro Linux (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Here is FreeCAD Link 
[image]

Version: 2021.504.24301 +3511 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: b8cd87c34a66a8f7a8177f483cebf79e29273813
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

QuakeString, 2021-05-09:
I'll Try Daily.
OS: Manjaro Linux (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.504.24301 +3511 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: b8cd87c34a66a8f7a8177f483cebf79e29273813
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)
[image]

QuakeString, 2021-05-09:
Result is quite similar.  
** Mistakenly I've "Closed" the issue.**

I've noticed one more thing, if I open my assembly and try to add my model, that time it take too long time and too much RAM, but if that model already in assembly and I open that assembly it uses 50%  of RAM.
[image]
[image]

realthunder, 2021-05-09:
You are testing using the same model as the OP right? The extra memory could be held in undo stack. My branch does use extra memory for topo naming. I'll investigate more.

QuakeString, 2021-05-09:
Yes, I've used exact same model. the same model I had problem with A2+ WB, it is now sorted after A2+ WB update.

realthunder, 2021-05-09:
What are the exact steps you did to experience the memory problem. I loaded up your `assembly_02` file, and the memory usage is 1.3GB. Then I click A2Plus edit assembly button to load the the `rotary_dryer_main_body_MOD02.FCStd`, it increases to 1.6GB. Then I click to reveal the hidden `DrumBody001`, it goes to 1.8GB.

Can you send me your `link.user.cfg` file located in directory `~/.FreeCAD`.

QuakeString, 2021-05-09:
here is link of files
https://drive.google.com/file/d/14frRh4dZwX5jazburxnE6j0P-lm0KLzT/view?usp=sharing
https://drive.google.com/file/d/1escskqqwatV1YTLm-kG8GUd_gAGBZ38g/view?usp=sharing

luzpaz, 2022-04-13:
@QuakeString can you zip the `link.user.cfg` file and attach it to this ticket? The google drive URLs are invalid now