realthunder/FreeCAD #134: Sketcher Feature Ideas

obelisk79, 2021-05-01:
Hello,

I have some feature requests/ideas I'd like to mention for consideration.

1. Give sketcher elements such as constraints, dimensional measurements or plane labels (XY, XZ, YZ) a 'bounding box' and/or improved layer precendence to make selection of elements easier.
  -The issue: is that sometimes the constraint/dimension is placed over other sketcher elements and the text (for example 22mm) will be entirely unselectable. I often have to delete a line to move dimensional constraints for clear reading in complex sketches. I also find it can be diffcult sometimes to select primary planes for starting a new sketch.
  
2. Modify layered drawing of elements to be normal to viewport perspective.
 -The issue: occasionally, when generating a quick model where parametric design isn't as critical and I create elements more directly, I like to work/sketch on the underside of a plane/body. Vertexes cannot be selected in this orientation. By modifying element layer behavior to work normal to the camera perspective this would alleviate this problem and potentially other aberrant behavior.
 
3. Add a setting which would highlight sketch elements that share a constraint by selecting the constraint icon in sketcher. Also highlighting of constraints associated with a line or vertex when element is selected.
   -The issue: In more detailed sketches, the list/display of constraints can become very cumbersome. By highliting attached constraints or elements when you select them can provide a good visual cue while working.
   
4. Add check-box to hide non-dimensional constraints from being displayed until an attached sketch element is selected.
 -The idea: visual clean-up when creating sketches with large numbers of constraints. This would only display constraint icons when you select an associated line, vertex or arc. This is particularly useful when zooming out and the visual representation of sketch becomes obfuscated by the constraint icons.
 
These are just some ideas I have had, based on some frustrations I have had over the last 6 months while using FreeCAD.

Thanks for your hardwork.

realthunder, 2021-05-01:
I do plan to do something about your second problem, but it's of low priority right now. For the rest of issues, please use the constraint list in the task panel. You can hide all constraint by clicking to top button as shown in the following screenshot, or toggle individual constraint visibility using the checkbox as shown below. You can use mouse over the item in the constraint list to highlight to help finding the corresponding constraint.

[image]

tiket18, 2021-05-04:
I like the author's idea - to provide a mode where all constraints are invisible, but when you select a particular geometry element = all constraints related to it become visible. Now a similar functionality is implemented by a separate button "show constraints related to the element" - but with hundreds of other constraints it is not realistic to cover the necessary ones in complex sketches. Let's say, we have hundreds of segments, six hundred and one is set as equal to the first and perpendicular to the fifteenth, symmetrical to something and tangent to something else - it's really hard to see everything.

luzpaz, 2022-04-13:
@obelisk79 can you summarize what's left in this ticket?