realthunder/FreeCAD #137: macos bundling questions ❓

ipatch, 2021-05-03:
hello, have a few questions related to macos.

1. are you providing updated builds for your active fork of freecad? if so where?

i glossed the README in this repo, but didn't see anything specific added based on changes related to this fork. i also looked at the releases within the releases tab of this repo and did not find any prebuilt pkgs for running this active fork of freecad.

2. i ended compiling this fork the other and built it with the below flag set to `ON` on my macos 10.14 mojave box.

i can successfully launch the app bundle `FreeCAD.app`, and it appears all workbenches are working, i can into sketcher create sketch, and then pad the sketch from the partdesign wb. and the draft wb loads without any noticable errors. 👍

however, i tried moving the app bundle `FreeCAD.app` from its install location to within my `/Applications` dir on macos, and then attempted to run the bundle/freecad by double clicking on it, and ran into some errors.

[image]

☝️ that is the error i'm seeing when i try to open the built app bundle from within the `/Applications` dir after copying the bundle from the specified install location after building freecad from source.

realthunder, 2021-05-03:
I release my prebuilt image in my [FreeCAD_Assembly3](https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.11) repo. But it is always good if you can build yourself. Are you running the application inside the build directory? Have you done `make install`? That should make the MacOS bundle relocatable. But since you've mentioned `FreeCAD.app`, you've probably already done that. Can you please click the `Report` button in the error dialog and post here the details about the error?

ipatch, 2021-05-04:
thanks for the clarification on where to download the prebuilt bins.

make adding a link to the readme within this repo may help those coming for external links and without prior knowledge on where to download your prebuilt binaries.

also i've been working with building the mac app bundled version of freecad from a couple of different forks recently, so i'm still experimenting with several different cmake args, so more than likely the error reported due to app bundling is more than likely self induced, but is the output from clicking that `Report` button.

```
Process:               FreeCAD [12721]
Path:                  /Applications/freecads/FreeCAD.my.rt.build.app/Contents/MacOS/FreeCAD
Identifier:            FreeCAD
Version:               ??? (0.19.16100)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           FreeCAD [12721]
User ID:               501

Date/Time:             2021-05-04 15:12:47.504 -0500
OS Version:            Mac OS X 10.14.6 (18G9028)
Report Version:        12
Anonymous UUID:        1D136EE9-33A2-5AA0-8415-2DDD37A831F2

Time Awake Since Boot: 530000 seconds

System Integrity Protection: disabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @rpath/libFreeCADGui.dylib
  Referenced from: /Applications/freecads/FreeCAD.my.rt.build.app/Contents/MacOS/FreeCAD
  Reason: image not found

Binary Images:
       0x10fecf000 -        0x10fedfff7 + (??? - 0.19.16100) <86204C19-9065-3AFB-9CA3-1D13AADF84F1> /Applications/freecads/FreeCAD.my.rt.build.app/Contents/MacOS/FreeCAD
       0x10feed000 -        0x10ffbdffb +libmpi.40.dylib (0) <C4FF07FF-6ADE-3B5E-A5DD-F424AAF24DF7> /usr/local/opt/open-mpi/lib/libmpi.40.dylib
       0x11e409000 -        0x11e47370f  dyld (655.1.1) <5C085E07-E51D-35EC-BE29-92A5782AA39F> /usr/lib/dyld

Model: MacBookPro11,3, BootROM MBP112.0145.B00, 4 processors, Intel Core i7, 2.3 GHz, 16 GB, SMC 2.19f12
Graphics: kHW_IntelIrisProItem, Intel Iris Pro, spdisplays_builtin
Graphics: kHW_NVidiaGeForceGTX750MItem, NVIDIA GeForce GT 750M, spdisplays_pcie_device, 2 GB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.77.61.3 AirPortDriverBrcmNIC-1305.12)
Bluetooth: Version 6.0.14d12, 3 services, 27 devices, 1 incoming serial ports
Network Service: 802.11ac NIC, Ethernet, en3
USB Device: USB 3.0 Bus
USB Device: 4-Port USB 3.0 Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: 4-Port USB 2.0 Hub
USB Device: USB Optical Mouse
USB Device: 802.11ac NIC
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1

ipatch, 2021-05-08:
adding some more info on mac app bundling. it appears the mac app bundle `FreeCAD.app` is assembled using a python 2 script as seen [here][ul1]. for whatever reason one of my deps is linking against ic4u version 69 instead of the 67 version provided by the freecad tap. and since i'm running mojave, i'll have to recompile qt (7+ hour build time on my box) and make sure it links against the proper version of icu4c. i copied above mentioned python script within the repo, and littered it with `print` statements to further see what my issue is, if you have any ideas on how i could further diagnose this issue i'd greatly appreciate your input.

and i'm aware this issue isn't really related to your fork, but more of upstream freecad issue, so please forgive if you see this msg as spam.

[ul1]: <https://github.com/FreeCAD/FreeCAD/blob/master/src/Tools/MakeMacBundleRelocatable.py>

realthunder, 2021-05-09:
The `icu` locating code is in `src/MacAppBundle/CMakeLists.txt:114`. As you can see, it finds the version through GLOB-ing, so it will have trouble if there are multiple version installed through `homebrew`. As a dirty workaround, I guess you can simply modify the CMakeLists to hardcode the correct version.

BTW, have you solved the above missing `libmpi` problem?