realthunder/FreeCAD #138: GUI - Critical problem of arbitrary rearrangement of elements

tiket18, 2021-05-04:
In all versions. Long time encountered, long time pissed off. But now I can show you on a simple example. There are segments, we drew them according to that scheme of mutual arrangement, as we need. Now we set the dimensions, setting the smallest now limit the total height of 10mm -
[image]

In response we get a change in the layout of the bottom right elements, which now cannot be changed (except only by deleting the size or element).

[image]
[ошибка.FCStd.zip](https://github.com/realthunder/FreeCAD/files/6419883/FCStd.zip)

So, I believe that such an arbitrary change in the layout of already limited geometry elements = error. On a simple example it's still easy to overcome, but in the case of complex sketches, and if this sketch is referenced by other elmenets = quite a problem.

realthunder, 2021-05-04:
I don't think this can be considered as a bug. It makes a difference with the order you apply those constraints. For example, it is generally [recommended](https://wiki.freecadweb.org/Sketcher_Workbench#Best_Practices) to apply for geometrical constraint (i.e. equality, coincidence, etc.) before dimensional constraint, but sometimes the other way round. By using the constraint list in the task panel, you can selectively activate/deactivate some constraints in order to adjust the geometry.

[image]

tiket18, 2021-05-04:
I couldn't agree more - I think it's a serious problem. The program does not change the size, but the arrangement of the elements. And this is critical, especially when these relocated elmenets are referenced in other thumbnails, etc. Cascading updates of all dependent elements in an avalanche-like manner, leads to uncontrollable destruction of more complex projects.

tiket18, 2021-05-04:
We cannot predict exactly how a change (setting) of a particular constraint (size) will affect (change in an unpredictable way) the arrangement of elements, so we cannot disable any size, constraint in advance ("would have known where to fall - would have made a straw bed in advance").

tiket18, 2021-05-04:
In fact, there is this logic in the operation of this phenomenon - when a size constraint (and \ or symmetry) is set, when the implementation of such a constraint requires moving that particular element to execute (implement) the constraint set, the program first moves exactly that element "individually by itself" and only then attracts adjoining elements to it. If, when moving it in this way, the element "oversteps" the other (adjacent) elements = their mutual positioning changes . Consequently, it is necessary to change the algorithm for moving the element.

tiket18, 2021-05-04:
Moreover, for this behavior (spontaneous rearrangement of elements) it is not even necessary to change the dimensions. You can create a sketch that has two degrees of freedom (e.g. a square and holes attached to its sides) and another sketch referring to this one - its elements and in this excise you can define everything (0 degrees of freedom). Then the first sketch, without changing its explicitly defined dimensions, just drag the mouse around the screen (in edit mode, all elements are linked and defined, move all together), exit. And we see that the dependent thumbnail has swapped some elements.

realthunder, 2021-05-05:
I think you are probably complaining about some deeper problem with the current sketch solver, which is out of my grasp at the moment. My advice is to find some way to cope with it, like creating a reference dimensional constraint first, and manually move the geometry to roughly what you have in mind. Then, apply geometrical constraints first (equality, symmetric, etc.). And finally turn those reference constraints to normal locking ones. 

https://user-images.githubusercontent.com/1207888/117094875-8819ef00-ad97-11eb-95b9-dc192d3827e3.mp4