realthunder/FreeCAD #139: strange behavior of the solver

tiket18, 2021-05-04:
It often happens so - we want to set a restriction (size, position, symmetry) - we try and the solver swears. Here's an example - a rectangle, in it drew horizontal and vertical lines (axes), I try to set the symmetry of points relative to the horizon = scolding -
[image]

 I take similar points relative to the same axis (to the left) = all is well.
[image]
[error_copy.FCStd.zip](https://github.com/realthunder/FreeCAD/files/6420662/error_copy.FCStd.zip)

luzpaz, 2022-01-18:
Please always attach your full [About](https://wiki/freecad.org/About) info.
Is this behavior you can reproduce on upstream as well?

tiket18, 2022-01-19:
Yes. The problem persists. In upsrim, too. In a constraint-rich sketch, quite often when you set the symmetry constraint and turn on the "automatically remove redundancies" option, FC turns on the "symmetry" requirement and does not remove the "vertical" or "horizontal" that becomes redundant.
That's not all :)
It also likes to cause a crash (undecidable, etc.) in a saturated sketch when trying to set a "point match" - when trying to close multiple arcs in a point. The workaround is to "put the point on the line".

dickshaydle, 2022-01-19:
If opened your file and the "Auto remove redundant" doesn't seem to work. It still says "partially redundant" even after pressing "update".

But i guess you could manually remove those redundant constraints and be fine.

tiket18, 2022-01-20:
If the solution you suggested had "worked", this ticket would not have been born. The problem is the partial operation of the constraint conflict solver. A typical disease is the symmetry restriction of the chosen axis, when the points from which symmetry is required are on a line with respect to which the verticality or horizontality condition was previously set. In this case the program specifies symmetry, but does not remove horizontality or verticality, wastes a lot of time searching for solutions and gives an abort. In saturated sketches "a lot of time" = a very long time.
Similar problem in trying to set matching points, I wrote above on the thread.

dickshaydle, 2022-01-20:
@tiket18 
It is still unclear what your problem is. I tried your file and in my opinion it works fine and as expected.
Your file is at the state before your first picture.
You added an unnecessary constrained of the vertical line in the middle of the box. That line was already constrained before.

The only problem i see is that the auto remove is not working as i would expect it.