realthunder/FreeCAD #141: Do you have plans for the next release?

leoheck, 2021-05-07:
Hi, @realthunder 

I am following your videos as soon as they are released, they are really good. Your work is awesome. 

I would like to know if you recommend your branch for use. I want to use it since it has some really nice features. Do you actually use this branch for your 3d models or you just write its code?

Ah, one more question, Do you have any plan for a new release? I am seeing here that the last one was a year ago.
And I could not find a list of features/fixes on each release. What do you think about adding it in next time, so people using it could actually use and test these changes?

Thank you.

obelisk79, 2021-05-07:
Releases found here:
https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.11

leoheck, 2021-05-07:
Oh, nice!

czinehuba, 2021-11-05:
@leoheck can you close this issue, or mark it done. Thanks

closed 2021-11-05 by leoheck