realthunder/FreeCAD #143: Ubuntu 21.04 compile fails

Kuzma30, 2021-05-09:
Ubuntu 21.04 Python 3.9.4 LinkDaily branch. Raspberry Pi4.

Scanning dependencies of target SketcherGui
[ 82%] Building CXX object src/Mod/Sketcher/Gui/CMakeFiles/SketcherGui.dir/CommandCreateGeo.cpp.o
[ 82%] Linking CXX shared library ../../../../Mod/TechDraw/TechDraw.so
/usr/bin/ld: CMakeFiles/TechDraw.dir/AppTechDraw.cpp.o: in function `PyInit_TechDraw':
/home/ales/BUILD/FreeCAD/src/Mod/TechDraw/App/AppTechDraw.cpp:118: undefined reference to `TechDraw::DrawTileWeld::init()'
/usr/bin/ld: /home/ales/BUILD/FreeCAD/src/Mod/TechDraw/App/AppTechDraw.cpp:144: undefined reference to `App::FeaturePythonT<TechDraw::DrawTileWeld>::init()'
/usr/bin/ld: CMakeFiles/TechDraw.dir/DrawWeldSymbol.cpp.o: in function `TechDraw::DrawWeldSymbol::onSettingDocument()':
/home/ales/BUILD/FreeCAD/src/Mod/TechDraw/App/DrawWeldSymbol.cpp:93: undefined reference to `typeinfo for TechDraw::DrawTileWeld'
/usr/bin/ld: /home/ales/BUILD/FreeCAD/src/Mod/TechDraw/App/DrawWeldSymbol.cpp:93: undefined reference to `typeinfo for TechDraw::DrawTileWeld'
/usr/bin/ld: /home/ales/BUILD/FreeCAD/src/Mod/TechDraw/App/DrawWeldSymbol.cpp:100: undefined reference to `typeinfo for TechDraw::DrawTileWeld'
/usr/bin/ld: /home/ales/BUILD/FreeCAD/src/Mod/TechDraw/App/DrawWeldSymbol.cpp:100: undefined reference to `typeinfo for TechDraw::DrawTileWeld'
/usr/bin/ld: CMakeFiles/TechDraw.dir/DrawWeldSymbol.cpp.o: in function `TechDraw::DrawWeldSymbol::getTiles() const':
/home/ales/BUILD/FreeCAD/src/Mod/TechDraw/App/DrawWeldSymbol.cpp:140: undefined reference to `TechDraw::DrawTileWeld::getClassTypeId()'
collect2: error: ld returned 1 exit status
make[2]: *** [src/Mod/TechDraw/App/CMakeFiles/TechDraw.dir/build.make:1535: Mod/TechDraw/TechDraw.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:5066: src/Mod/TechDraw/App/CMakeFiles/TechDraw.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs...

projectcad, 2021-05-14:
Same error at my build too

realthunder, 2021-05-14:
This looks like a parallel make problem. Can simply try make again?

luzpaz, 2022-01-18:
@Kuzma30 bump

closed 2022-04-20 by Kuzma30