realthunder/FreeCAD #144: Hovered part rendering transparency issue (on Wayland)

unrelentingtech, 2021-05-09:
The renderer is changing framebuffer alpha (for the parts under the mouse cursor), expecting it to be ignored (even though nothing in the code explicitly only allows framebuffers with no alpha), similar to https://github.com/Ultimaker/Cura/issues/8466 https://stackoverflow.com/q/35245809

[image]

(the right side actually looks fully yellow on screen, the glitches are only on screenshots… and the fact that the behavior is different on the left and right sides of the screen suggests some kind of HiDPI issue somewhere?)

realthunder, 2021-05-16:
This looks a difficult problem to solve as it is highly platform specific. Meanwhile, just in case you don't know, you can disable `Preselection highlight`, `Preselection on top`, and `Selection on top` to prevent the object goes transparent.

[image]

[image]

unrelentingtech, 2021-05-16:
I still haven't figured out how to do it in Qt, but generally it's trivial to request a framebuffer without alpha. e.g. https://github.com/aap/librw/pull/35 in GLFW it's just `glfwWindowHint(GLFW_ALPHA_BITS, 0)` and that's it

unrelentingtech, 2021-09-03:
Seems fixed right now in LinkDaily (or it wasn't FreeCAD's fault in the first place).

closed 2021-09-03 by unrelentingtech