realthunder/FreeCAD #145: Ubuntu 18.04 building failed..

projectcad, 2021-05-13:
Please be informed that after updating linkstage3 of today...it returns error like below while building..
[image]

realthunder, 2021-05-14:
Can please find and delete the file EditDatumDialog.cpp.o and try make again?

projectcad, 2021-05-14:
Thanks.. Building is OK..
but this time, another error like below while installing with "sudo make install"
[image]

realthunder, 2021-05-14:
Well, I guess you'll have to find and delete `MeshPartGui.so` in your build directory (note, not `/usr/local/lib`), and `make install` again. I think there is some problem with Linux parallel make. Maybe some race condition somewhere.

projectcad, 2021-05-14:
👍Finally OK...
after deleting MeshPartGui.so 
i tried building  and installing again..
no any error at both of them.. moreover, no any problem at EditDatumDialog.cpp.o and MeshPartGui.so (before deleting them, their sizes are 0...)

closed 2021-05-14 by projectcad