realthunder/FreeCAD #149: FreeCad 2021.519.24301 +3652 crash

hdt551, 2021-05-20:
1) Create new empty document.
2) Create body.
3) Create sketch. (for me FreeCad program crashes here)

OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.519.24301 +3652 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 766fd735b25889fecffcebe42b9d69bd24cf4006
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

realthunder, 2021-05-20:
Can you please launch the AppImage from a terminal? When the crash happen, it will usually print out the crashing call stack trace. Please copy and paste the trace here. Also, please attach your `link.user.cfg` file here, in `~/.FreeCAD` directory.

hdt551, 2021-05-20:
Please find requested files attached.

[link_user_cfg_and_segfault.zip](https://github.com/realthunder/FreeCAD/files/6516395/link_user_cfg_and_segfault.zip)

hdt551, 2021-05-28:
I can confirm this is fixed in version 2021.528.24301 +3686 (Git) AppImage

closed 2021-05-28 by hdt551