realthunder/FreeCAD #150: Compiling of LinkDaily Branch returns error at Mesh.dMesh.dir

projectcad, 2021-06-06:
Compiling of LinkDaily branch returns error...i tried also "make" command with single core..but same error
[image]
[image]
[image]
[image]
[image]

realthunder, 2021-06-07:
The error doesn't make much sense. Probably another weird problem caused by parallel build. Please manually delete the `Mod/Mesh` directory in your **build** directory and try again.

projectcad, 2021-06-08:
Thanks..it is OK after deleting Mesh dir...but very interesting that paralel compiling causes error only at Mesh...no any error at others..

closed 2021-06-08 by projectcad

projectcad, 2021-06-11:
Updated and then tried to compile with paralel cores today...no any error this time..