realthunder/FreeCAD #153: Problems with Pocket and Fit tolerance

jx2014, 2021-06-24:
Steps to recreate the issue:
1. create a sub-object shape binder
2. pocket the shape binder
3. enter any non-zero values into 'Fit tolerance'.
4. the expansion appears to be in the wrong direction, see example below:

Full tutorial here:
[https://www.youtube.com/watch?v=OWNrYvxpG4k&t=720s](url)

Expected result (as shown in the tutorial in Apr 23, 2021, Linkstage 3 Build unknown.):
[image]

Actual result: 
The expansion appears to be in the wrong direction.
[image]

FCStd file:
[1.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6707404/1.zip)

> OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3584 (Git)
Build type: Release
Branch: LinkStage3
Hash: d1e4d1c5abdedda669e54d1a41b5c08077f7e561
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

realthunder, 2021-06-26:
This Pad/Pocket direction problem has been fixed in cf273a74dd255f7de5f57e062b556c6db80980c5. However, you pocket through shape binder still can't work. It turns out that your imported usb module shape is invalid. You can select that shape, and run `Part -> Check geometry` to see lots of errors. The bound face got error as well. I am not sure what causes this. Can I have the original STEP file?

jx2014, 2021-06-27:
> This Pad/Pocket direction problem has been fixed in [cf273a7](https://github.com/realthunder/FreeCAD/commit/cf273a74dd255f7de5f57e062b556c6db80980c5). However, you pocket through shape binder still can't work. It turns out that your imported usb module shape is invalid. You can select that shape, and run `Part -> Check geometry` to see lots of errors. The bound face got error as well. I am not sure what causes this. Can I have the original STEP file?

Hi, the original STEP file comes from here: [https://grabcad.com/library/raspberry-pi-27](url)
I found a work around which was to create a draft, and use the Sketch External Geometry tool to trace the outline of the profile.

jx2014, 2021-07-19:
verified and it is fixed in in 2021.7.17 version:

>OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3627 (Git)
Build type: Release
Branch: LinkStage3
Hash: f63a5ea7b459872a3798027df0ff25e776524859
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

closed 2021-07-19 by jx2014