realthunder/FreeCAD #155: Toggling the visibility of padded object has no effect to its child sketch

jx2014, 2021-07-01:
steps to reproduce the issue:

1. create a new part
2. create a new body
3. create a new sketch
4. pad the sketch
5. [image]
5. toggle the visibility of the part, everything disappears, this is expected.
6. toggle the visibility of the body, everything disappears, this is expected.
7. [image]
7. toggle the visibility of the pad, the sketch remains visible, this is unexpected.
8. [image]
9. why is this a problem? sometimes it can create confusions when there are multiple padded objects, in which case one needs to manually toggle the visibility of each sketch. 
10. [image]

> OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3584 (Git)
Build type: Release
Branch: LinkStage3
Hash: d1e4d1c5abdedda669e54d1a41b5c08077f7e561
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

Rabbit-sk, 2021-07-15:
When you create a pad from a sketch, the sketch visibility is automatically turned off.
So if you turn off the visibility of the pad, the sketch is also not visible, because it was already not visible.

> toggle the visibility of the pad, the sketch remains visible, this is unexpected.

You can toggle the visibility of solid features (pads, pockets, ...), and their child non-solid features (sketches, ShapeBinders, ...), independently  and it is not a bug but a feature. 

Many times it is convenient to turn off the visibility of the solid shape in a body in order to see the particular sketch. For example, when creating external geometry in a different sketch in the same body. Also for simple selection of particular edges when creating SubShapeBinder in another body, as well as in other cases that I don't remember now.

luzpaz, 2022-01-18:
Closing ticket

closed 2022-01-18 by luzpaz