realthunder/FreeCAD #159: Windows 10 build fails. No boost lib found

Kuzma30, 2021-07-22:
When I try to configure with Cmake 3.8.2 I get such error
https://pastebin.com/czxQNBNj

I use FreeCADLibs_asm3_12.5.5_x64_VC17.7z and latest daily branch.

realthunder, 2021-07-23:
I can't access the pastebin link (404 error). Can please you post again?

Kuzma30, 2021-07-24:
https://pastebin.com/A56NaA1t

Kuzma30, 2021-07-24:
https://pastebin.pl/view/885e6d4e

realthunder, 2021-07-25:
What is your cmake command? It looks like you didn't specify the libpack directory. This is mine.

```bat
cmake -G "Visual Studio 16 2019" -A  x64 ^
            -DFREECAD_LIBPACK_DIR="%libpackpath%" ^
            -DFREECAD_USE_PYBIND11:BOOL=ON ^
            -DBUILD_FLAT_MESH:BOOL=ON ^
            -DBUILD_FEM_NETGEN:BOOL=ON ^
            -DOCC_INCLUDE_DIR="%libpackpath%/include/opencascade" ^
            -DPYTHON_EXECUTABLE="%libpackpath%/bin/python.exe" ^
            ..
```

Kuzma30, 2021-07-25:
Run cmake with your command
https://pastebin.pl/view/e3922c7e

realthunder, 2021-07-25:
I didn't have `-DBOOST_LIBRARYDIR` argument. Also, please use a newer cmake. I am using 3.14.5.

Kuzma30, 2021-07-25:
I now use latest cmake, it configure fine. Thanks.

luzpaz, 2022-01-18:
@Kuzma30 pls close ticket

realthunder, 2022-01-18:
@luzpaz I have just added you as the collaborator of this repo. Did you get the invitation?

closed 2022-01-18 by realthunder