realthunder/FreeCAD #160: Faield build on Manjaro

JFincher42, 2021-07-23:
Getting the same persistent error trying to build from https://github.com/realthunder/FreeCAD/tree/LinkStage3 on Manjaro:

> /usr/include/boost/geometry/index/rtree.hpp:285:21: error: using invalid field ‘boost::geometry::index::rtree<Value, Options, IndexableGetter, EqualTo, Allocator>::members_holder::root’
> make[2]: *** [src/Mod/Sketcher/App/CMakeFiles/Sketcher.dir/build.make:204: src/Mod/Sketcher/App/CMakeFiles/Sketcher.dir/SketchObject.cpp.o] Error 1
> make[1]: *** [CMakeFiles/Makefile2:3178: src/Mod/Sketcher/App/CMakeFiles/Sketcher.dir/all] Error 2
> make: *** [Makefile:136: all] Error 2

This is after many _many_ errors all complaining about using invalid fields in boost. I followed the steps at https://wiki.freecadweb.org/Compile_on_Linux#Out-of-source_build for an out of source build, after cloning the LinkStage3 repo locally.

Boost version is boost-1.76.0-1, gcc is gcc-11.1.0-1.

Manjaro details:

> ❯ neofetch
██████████████████  ████████   jon@xxxxxx 
██████████████████  ████████   ------------ 
██████████████████  ████████   OS: Manjaro Linux x86_64 
██████████████████  ████████   Host: XPS 8930 1.1.13 
████████            ████████   Kernel: 5.13.1-3-MANJARO 
████████  ████████  ████████   Uptime: 9 days, 3 hours, 23 mins 
████████  ████████  ████████   Packages: 1879 (pacman) 
████████  ████████  ████████   Shell: zsh 5.8 
████████  ████████  ████████   Resolution: 1920x1080 
████████  ████████  ████████   DE: Plasma 5.22.3 
████████  ████████  ████████   WM: KWin 
████████  ████████  ████████   Theme: Breeze Dark [Plasma], Breath-Dark [GTK2/3] 
████████  ████████  ████████   Icons: breeze-dark [Plasma], breeze-dark [GTK2/3] 
████████  ████████  ████████   Terminal: yakuake 
                               CPU: Intel i7-9700 (8) @ 4.700GHz 
                               GPU: AMD ATI Radeon RX 5600 OEM/5600 XT / 5700/5700 XT 
                               GPU: Intel CoffeeLake-S GT2 [UHD Graphics 630] 
                               Memory: 10968MiB / 31819MiB

realthunder, 2021-07-23:
It could be the same problem mentioned in [this](https://forum.freecadweb.org/viewtopic.php?t=47797) post. Please check the boost fix [here](https://github.com/boostorg/geometry/commit/a74a2b5814a8753013a8966606b8472178fffd14) and see if it applies to you.

JFincher42, 2021-07-24:
That fix is already on my system, so this isn't it.

I've attached the captured output and errors from the `make` command, if it helps.

[errors.txt](https://github.com/realthunder/FreeCAD/files/6872506/errors.txt)
[output.txt](https://github.com/realthunder/FreeCAD/files/6872507/output.txt)

realthunder, 2021-07-24:
It's probably your `cmake` arguments need some tuning. Here is mine for your reference. Some of the errors seems related to not enabling C++11 build option.

```
cmake \
    -DOCCT_CMAKE_FALLBACK=1 \
    -DFREECAD_USE_OCC_VARIANT="Official Version" \
    -DBUILD_QT5:BOOL=ON \
    -DUSE_BOOST_PYTHON:BOOL=OFF \
    -DFREECAD_USE_PYBIND11:BOOL=ON \
    -DBUILD_ENABLE_CXX11:BOOL=ON \
    -DPYTHON_EXECUTABLE:FILEPATH=/usr/bin/python3 \
    -DBUILD_FEM_NETGEN:BOOL=ON \
    -DBUILD_PLOT:BOOL=OFF \
    -DBUILD_SHIP:BOOL=OFF \
    -DFREECAD_USE_QT_DIALOG:BOOL=ON \
    -DBoost_NO_BOOST_CMAKE:BOOL=ON \
    -DFREECAD_USE_QWEBKIT:BOOL=ON \
    -DUSE_BOOST_PYTHON:BOOL=OFF \
    -DBUILD_FLAT_MESH:BOOL=ON \
    ../
```

JFincher42, 2021-07-24:
No joy there, either - same error. In fact, I had to remove the `-DBUILD_FEM_NETGEN:BOOL=ON` line or CMake would throw errors and not finish properly.

I tried it on my laptop (which is Manjaro as well), but no luck there either - same errors. I'm thinking there is some mismatch with the Boost headers and the libs, but I can't find anyone else having this problem with anything else.

I will try to compile the baseline FreeCAD to see if the problem is there - if so, I will redirect this to them.

JFincher42, 2021-07-25:
FWIW, I can build the baseline FreeCAD with no issues. I'll dig in to see where my mismatches might be.

antonsb, 2021-07-30:
i have the same issue on Manjaro:
gcc (GCC) 11.1.0
boost 1.76.0-1

wohltat, 2021-08-23:
Have you tried Ceremcems build environment https://github.com/ceremcem/build-freecad-asm3 ?
I find this a more reproducible way of compiling.

luzpaz, 2022-01-18:
Status of this ticket?

closed 2022-01-18 by luzpaz