realthunder/FreeCAD #163: [Feature request] Custom direction for pad and pockets relative to the normal

WiseroOrb, 2021-08-19:
I think a good enhancement for the pad/pocket feature is to define a custom direction **_t_** relative to the normal **_n_**,
given 2 other angles, like in this image.

[image]

wohltat, 2021-08-21:
you can already do that with the extrude tool
[image]

WiseroOrb, 2021-08-21:
Not quite, because in the extrude tool you specify a global direction, relative to the origin.
Instead I'm suggesting to take the normal of the face and angle it with 2 parameters.

wohltat, 2021-08-21:
you could set the x, y, z of the custom direction as you like.
here is an example for the normal of a box:
```
x:
Box.<<Face6>>._shape.normalAt(0, 0)[0]
y:
Box.<<Face6>>._shape.normalAt(0, 0)[1]
z:
Box.<<Face6>>._shape.normalAt(0, 0)[2]
```

It is missing a point and click interface but if you are willing to use some math and python you should be able to do everything with it.
Once you press "=" or the "f(x)" button to enter the expression, it should be easy to find the face because of the autocompletion.
The name of the face can be seen in status bar or in the python console if you select it.

For some unknown reason there sometimes is an error message. But it disappeared as soon as i rotated the cube.

```
12:21:39  <string>(1)<class 'SystemError'>: <built-in method normalAt of Part.Face object at 0x558ca34e24a8> returned a result with an error set
in expression: Box.<<Face6>>._shape.normalAt(0, 0)[0]
in binding '.Direction.x'
12:21:39  Recompute failed! Please check report view.
```

WiseroOrb, 2021-08-21:
Interesting,  I didn't know I can get the normal in that way.
I'm going to think for a suitable algorithm. 
Keep this issue open, I will post a solution if I find one

wohltat, 2021-08-21:
I tried to add a local coordinate system (icon below datum plane) and rotate it.
```
Local_CS.Shape.valueAt(0, 0)[0]
...
```

In the preview it works fine but if i click ok there is always a similar error as before that i don't know how to work around.

```
14:15:58  <string>(1)<class 'SystemError'>: <built-in method valueAt of Part.Face object at 0x558ca6359ed8> returned a result with an error set
in expression: Local_CS.Shape.valueAt(0, 0)[0]
in binding '.Direction.x'
14:15:58  Recompute failed! Please check report view.
```

WiseroOrb, 2021-08-23:
Ok, I've put together some math, however I need help to integrate the python code into Freecad. I've never done it.

Math:

n : face normal
up : a given vector to define the up direction inside of the plane of the extrusion
It's defined by a referenced geometry 
U = (U - n*(n*up)).normalize : extract the component orthogonal to n and normalize it
For the math to work up can't be parallel to n
R = Uxn : cross product to find the last vector to complete the local coordinate system
LCS: (R,U,n) : (x', y', z')

Now given two angles alpha and beta and extrusion length L, I calculate the extrusion vector

x' = L\*sin(beta)\*sin(alpha) 
y' = L\*cos(beta)\*sin(alpha) 
z' = L*cos(alpha)

Now I calculate the global extrusion vector

(x, y, z) = R\*x' + U\*y' + n*z'

The python code is straightforward to implement, but as I don't know how to integrate it into freecad

realthunder, 2021-08-23:
> In the preview it works fine but if i click ok there is always a similar error as before that i don't know how to work around.

There is a bug in expression evaluation causing this hard to understand error, which has already been fixed. There is actual syntax error in your expression, but the bug is masking the syntax error. I am guessing you didn't put a space after the comma in function argument. The expression allows to use comma as fraction dot as in some locale. So, you must include a space after the comma for disambiguation.

> Ok, I've put together some math, however I need help to integrate the python code into Freecad. I've never done it.

The math can be simplified assuming you know the concept of [euler angles](https://en.wikipedia.org/wiki/Euler_angles#Conventions_by_extrinsic_rotations). The operation you want is an orientation represented by extrinsic euler angles in order of either XY or YX. Meaning rotate some angle by X axis and another angle by Y axis. Extrinsic means, the rotation axis is static. The simplify things a bit more, you can attach a `Local coordinate system` to the face, assuming its named `Local_CS`. You can manually adjust `Local_CS` attachment offset to fine tune the direction. You can enable extrusion by custom direction and then bind the following expression directly to the `Direction` property of the `Pad` feature in the property view (right click the property and choose `Expression...`),

```
Local_CS.Placement.Rotation * ._app.Rotation(<<XYZ>>, 20, 30, 0) * ._app.Vector(0, 0, 1)
```

This is equivalent to the following Python code

```
App.ActiveDocument.Local_CS.Placement.Rotation * App.Rotation('XYZ', 20, 30, 0) * App.Vector(0, 0, 1)
```

The particular constructor of `Rotation` creates an orientation using extrinsic rotation in order X 20 degree, Y 30 degree, and Z 0 degree. You can also do intrinsic angle with something like 'IXYZ'.

To make it even more parametric, you can add separate property for X and Y rotation angles in the `Pad` feature (right click in property view, and choose `Show all`, right click again, and choose `Add property`). Then refer to those properties in expression.

wohltat, 2021-08-23:
Works great. I didn't know one can create properties, pretty cool.
[image]

> There is a bug in expression evaluation causing this hard to understand error, which has already been fixed. There is actual syntax error in your expression, but the bug is masking the syntax error. [...]

It wasn't about the space because there already was one. You can see that in the copied error message. I also noticed and wondered that there has to be a space in the parameter list and guessed it had to do with locale stuff because comma is used as a decimal separator on my system. If the comma is missing there is another error message.
The strange thing is that is works in the evaluation window most of the time but not always but later there is always an error message.

[image]

WiseroOrb, 2021-10-03:
> > It's possible to add custom direction to the pocket?
> 
> In the next release, almost all `PartDesign` feature will have an extra option in editing panel for selecting the desired operation from`Additive`, `Subtractive`, to `Intersecting`. So you can easily use a `Pad` for pocketing.

That's a great way to combine those features. Looking forward to it.

closed 2021-10-03 by WiseroOrb