realthunder/FreeCAD #168: Transparent overlay not working with breeze application style (Linux/KDE)

frogmaster, 2021-08-22:
OS: Arch Linux
DE: KDE
freecad version: linkstage3 branch

Problem description: When using breeze application style (kde default), majority of panel (everything except panel titlebar) is not transparent.

[image]

Workaround: Panel transparency works as expected, when setting style to fusion in qt5ct and running freecad with "QT_QPA_PLATFORMTHEME=qt5ct freecad". Although the undesired side effect of this is out of place look compared to other applications, as the widget style is different.

unrelentingtech, 2021-09-03:
Same issue with [Adwaita](https://github.com/FedoraQt/adwaita-qt) style.

hellst0rm, 2021-10-15:
This workaround combined with choosing gradient background (both side of the gradient set the same) I was able to have a solid background and use the transparent overlays.

unrelentingtech, 2021-10-15:
oh, #144. Not nvidia related, I've had it on mesa/amd. It somehow stopped being a problem for me at some point, possibly with a LinkDaily update (?)

rmu75, 2022-10-20:
I have the same issue. Transparent overlay not working with breeze theme -- no difference between native wayland and Xwayland. It seems to work with fusion, ms9x or oxygen theme.  Latest AppImage does work. I probably could debug this but I have no idea where to start.

```
OS: KDE neon User - 5.25 (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.20.31486 (Git)
Build type: Debug
Branch: LinkDaily
Hash: a3d793869825443b46d981707dbdbe8b8d72b168
Python version: 3.8.10
Qt version: 5.15.6
Coin version: 4.0.0
OCC version: 7.5.2
Locale: German/Austria (de_AT)
```