realthunder/FreeCAD #173: LinkDaily build fails on ubuntu 20.04

wsteffe, 2021-09-04:
While LinkStage3 was successfully built on the same machine (ubuntu 20.04) without problems, the compilation of LinkDaily terminated with the following error message:

Scanning dependencies of target PathGui
[ 89%] Building CXX object src/Mod/Path/Gui/CMakeFiles/PathGui.dir/qrc_Path.cpp.o
[ 89%] Building CXX object src/Mod/Path/Gui/CMakeFiles/PathGui.dir/Command.cpp.o
[ 89%] Building CXX object src/Mod/Path/Gui/CMakeFiles/PathGui.dir/AppPathGui.cpp.o
[ 89%] Building CXX object src/Mod/Path/Gui/CMakeFiles/PathGui.dir/AppPathGuiPy.cpp.o
[ 89%] Building CXX object src/Mod/Path/Gui/CMakeFiles/PathGui.dir/PreCompiled.cpp.o
[ 89%] Building CXX object src/Mod/Path/Gui/CMakeFiles/PathGui.dir/DlgSettingsPathColor.cpp.o
/home/walter/Software/FreeCad/FreeCAD/src/Mod/Path/Gui/DlgSettingsPathColor.cpp:101:10: fatal error: moc_DlgSettingsPathColor.cpp: No such file or directory
101 | #include "moc_DlgSettingsPathColor.cpp"
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

realthunder, 2021-09-04:
This problem has been solved recently. Please sync the branch and try again.

wsteffe, 2021-09-04:
OK. Synced and successfully built.  Problem solved.

closed 2021-09-04 by wsteffe