realthunder/FreeCAD #174: LinkDaily Branch: Sketch doesn't work with parametric dimensions

wsteffe, 2021-09-05:
The annexed FC file is a simple structure consisting of  an extruded sketch.
[CMP_SFILT_6_4.zip](https://github.com/realthunder/FreeCAD/files/7110984/CMP_SFILT_6_4.zip)
The sketch failed to regenerate after I have redefined a few related dimensions using spreadsheet variables.
You may see that the sketch regeneration works again if you enter numeric values for those dimensions
(replacing the spreadsheet parameters).

wsteffe, 2021-09-05:
I suspect that this problem arises because the spreadsheet comes after the sketch in the history tree.

So I wanted to relocate (or redefine)  the spreadsheet before the sketch.
But here I discovered another problem:
 1) Right clicking Selecting Pad or Sketch in the annexed document there appears a command named "move object after another object" but it is grayed out.
 2) Somewhere I have seen that FC should provide a command  to "set tip" on a object in the model tree but this command seems not to be available in Linkdaily. 

So I can not find a way to put the spreadsheet before the Pad/Sketch other than delete all end create a new document.

I would also say that if spreadsheet variables must be defined before a sketch using them, then the user should non be allowed to refer to an after defined spreadsheet when reediting the sketch.

realthunder, 2021-09-05:
The item in the tree does not reflect the recomputation order. FreeCAD will auto sort the objects based on their dependency order before recompute. The problem you are facing is most likely a sketcher solver problem. I find that '8' figure wire is not closed for some reason. Printing each vertex of its edge reveal that there are small distances between some edges. For example,

```
>>> App.ActiveDocument.Sketch.Shape.Wire2.Edge1.Vertex2.Point -  App.ActiveDocument.Sketch.Shape.Wire2.Edge4.Vertex2.Point
Vector (-1.9960675476937695e-05, 3.154537229743681e-05, 0.0)
```

The default OCC tolerance is 1e-7. So a distance of 1e-5 is considered not connected. If you shake your sketch a abit, it will eventually make the connection. I suggest you report this case to the forum, as I can reproduce the problem in upstream 0.19 (Haven't tried with 0.20 yet). You'd better delete the Part container as it has delayed origin creation which is only available in my branch.

wsteffe, 2021-09-05:
This issue is also reported at https://forum.freecadweb.org/viewtopic.php?f=20&t=61944.
I have just added a new post and annexed a simplified FC document compatible with upstream FC which reproduces this sketch issue.

wsteffe, 2021-09-06:
I have just understood that this problem was due to my bad definition of sketch constraints. It is not a FC issue.

closed 2021-09-06 by wsteffe