realthunder/FreeCAD #175: just a question

wsteffe, 2021-09-06:
is it possible to put the spreadsheet into an external file (csv or xlsx) so that I can write it with an external code in order to change the CAD geometry ? I would like to extend an iterative optimization procedure (already working with Catia and ZW3D) so that it can interact also with FC.
I am not sure if this is the write place for this kind of question. So please let me know if you prefer that I readdress it to a different place.

wsteffe, 2021-09-07:
In my opinion it were very useful if the import command of the spreadsheet WB could be extended so that it keeps a link to the imported csv file and the spreadsheet values are updated (in a recompute) if they are changed in the csv file. The spreadsheet update should not modify any other data such as, in example, an alias which the user may have added after having imported the csv file.

wohltat, 2021-09-23:
[image]

You can also create a Macro with the following code. This will overwrite the first spreadsheet with the file specified in the `csv_filename` property which is created when you run the script the first time. You need to save the csv with tabs as delimiters which is default for FreeCAD (if you import via the Spreadsheet import/export button).

``` python
import csv

# use the first spreadsheet in the document
sheet = App.ActiveDocument.findObjects(Type='Spreadsheet::Sheet')[0]

# create a property to store the csv filename in
if not hasattr(sheet, 'csv_filename'):
    sheet.addProperty('App::PropertyString', 'csv_filename')

with open(sheet.csv_filename, 'r') as csv_file:
  # sheet.clearAll() # clearAll also clears property so just clear some range of cells
  sheet.clear('A1:Z1024')
  reader = csv.reader(csv_file, delimiter='\t')
  for rownum, row in enumerate(reader):
    for colnum, cell_content in enumerate(row):
      cell_address = chr(0x41 + colnum) + str(rownum+1)
      # print(cell_address, ': ', cell_content, end='\t')
      sheet.set(cell_address, cell_content)
    # print()

sheet.recompute()
```

@wsteffe  Btw, could you please rename the issue to something more meaningful than "just a question" or "how to do stuff"? ;)

wsteffe, 2021-09-26:
Thanks to all, I had forgotten this post.
Indeed I already (partially) solved my problem using a macro as suggested by wholtat. The macro (here annexed) [FCvarUpdate.zip](https://github.com/realthunder/FreeCAD/files/7232259/FCvarUpdate.zip)
was written with the help of TheMarkster in the post https://forum.freecadweb.org/viewtopic.php?f=3&t=61990&hilit=wsteffe.

The solution proposed by ceramcem is also interesting. 

I do not know how to rename this issue. RT, If you want, you may close it.

But perhaps it would be nice to have a predefined user command (not a user defined macro) to dynamically link the internal variables (which in FC are stored in the internal spreadsheets) to an external file. This kind of functionality is provided by most commercial CADs

wsteffe, 2021-09-26:
Ah, it was very easy to rename it.