realthunder/FreeCAD #176: Working in Path WB I recieved a: Program received signal SIGSEGV, Segmentation fault.

snelweg, 2021-09-09:
`Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libpthread.so.0(+0x12730) [0x7f15d11fe730]
#1  0x7f15d0aa9e46 in QLineEdit::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x26
#2  /usr/lib/python3/dist-packages/PySide2/QtWidgets.cpython-37m-x86_64-linux-gnu.so(+0x425703) [0x7f15a7b14703]
#3  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyMethodDef_RawFastCallKeywords+0x2e8) [0x7f15d1f53cb8]
#4  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyMethodDescr_FastCallKeywords+0x4a) [0x7f15d1f5480a]
#5  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalFrameDefault+0x8f3c) [0x7f15d1d56d5c]
#6  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x794d3) [0x7f15d1d584d3]
#7  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalFrameDefault+0x9334) [0x7f15d1d57154]
#8  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalCodeWithName+0x996) [0x7f15d1e7d1e6]
#9  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyFunction_FastCallDict+0xae) [0x7f15d1f5430e]
#10  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyObject_Call_Prepend+0xcd) [0x7f15d1f5720d]
#11  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x215af5) [0x7f15d1ef4af5]
#12  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x21f072) [0x7f15d1efe072]
#13  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyObject_FastCallKeywords+0xd3) [0x7f15d1f55043]
#14  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalFrameDefault+0x612e) [0x7f15d1d53f4e]
#15  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x794d3) [0x7f15d1d584d3]
#16  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyFunction_FastCallDict+0x2e2) [0x7f15d1f54542]
#17  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyObject_Call_Prepend+0xcd) [0x7f15d1f5720d]
#18  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x215af5) [0x7f15d1ef4af5]
#19  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x21f072) [0x7f15d1efe072]
#20  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyObject_FastCallKeywords+0xd3) [0x7f15d1f55043]
#21  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalFrameDefault+0x52eb) [0x7f15d1d5310b]
#22  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalCodeWithName+0x996) [0x7f15d1e7d1e6]
#23  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyFunction_FastCallKeywords+0x93) [0x7f15d1f54123]
#24  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalFrameDefault+0x9334) [0x7f15d1d57154]
#25  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalCodeWithName+0x996) [0x7f15d1e7d1e6]
#26  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyFunction_FastCallDict+0xae) [0x7f15d1f5430e]
#27  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyObject_Call_Prepend+0xcd) [0x7f15d1f5720d]
#28  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(PyObject_Call+0x75) [0x7f15d1f54c75]
#29  0x7f15d53f10fd in Base::pyCall(_object*, _object*) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x28
#30  0x7f15d5a820dc in Gui::ViewProviderPythonFeatureImp::setEdit(int) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x2d2
#31  0x7f15927fb18b in Gui::ViewProviderPythonFeatureT<PathGui::ViewProviderPath>::setEdit(int) from /home/ton/FREECAD/daily/freecad/Mod/Path/PathGui.so+0x27
#32  0x7f15d5a2c9f5 in Gui::ViewProvider::startEditing(int) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x2d
#33  0x7f15d5a5ff9f in Gui::ViewProviderDragger::startEditing(int) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x2f
#34  0x7f15d536960e in Gui::Document::setEdit(Gui::ViewProvider*, int, char const*) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x142a
#35  0x7f15927f7ed5 in PathGui::ViewProviderPath::doubleClicked() from /home/ton/FREECAD/daily/freecad/Mod/Path/PathGui.so+0x37
#36  0x7f15927fb545 in Gui::ViewProviderPythonFeatureT<PathGui::ViewProviderPath>::doubleClicked() from /home/ton/FREECAD/daily/freecad/Mod/Path/PathGui.so+0x45
#37  0x7f15d5706109 in Gui::TreeWidget::onDoubleClickItem(QTreeWidgetItem*) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x2fb
#38  0x7f15d5705ddc in Gui::TreeWidget::mouseDoubleClickEvent(QMouseEvent*) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x52
#39  0x7f15d09af00f in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xd0f
#40  0x7f15d0a51cde in QFrame::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1e
#41  0x7f15d0bc31eb in QAbstractItemView::viewportEvent(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1ab
#42  0x7f15d0c2b3eb in QTreeView::viewportEvent(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x3b
#43  0x7f15cfff519b in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x9b
#44  0x7f15d09704b1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x71
#45  0x7f15d0977bb8 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x458
#46  0x7f15d5424259 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0xff
#47  0x7f15cfff5489 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x179
#48  0x7f15d0976ee9 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1a9
#49  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1b7fd4) [0x7f15d09c8fd4]
#50  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1bab5e) [0x7f15d09cbb5e]
#51  0x7f15d09704c1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x81
#52  0x7f15d0977970 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x210
#53  0x7f15d5424259 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0xff
#54  0x7f15cfff5489 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x179
#55  0x7f15d03a0fa1 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x681
#56  0x7f15d03a2e85 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x135
#57  0x7f15d037d07b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xab
#58  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0xdd49b) [0x7f15c676149b]
#59  0x7f15cfff415b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13b
#60  0x7f15cfffc132 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x92
#61  0x7f15d52dbb00 in Gui::Application::runApplication() from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x5f2
#62  /home/ton/FREECAD/daily/freecad/bin/FreeCAD(+0x9168) [0x563c7cebf168]
#63  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7f15cf86009b]
#64  /home/ton/FREECAD/daily/freecad/bin/FreeCAD(+0x84da) [0x563c7cebe4da]`

> OS: Debian GNU/Linux 10 (buster) (X-Cinnamon/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.28391 (Git)
Build type: Unknown
Branch: LinkDaily
Hash: 50983c9c8f56409415833d495514aaf172d18358
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

snelweg, 2021-09-10:
Getting weird results in Path workbench and loads of crashes

`<Path.Area> PathAreaOp.py(206)|Area.cpp(1438): hit bottom 0,0,1e-06
Pocketing Select Mode
Please select faces from a single solid
Free Select
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x37840) [0x7f6f34ab7840]
#1  0x7f6f35bdaccf in QWidget::windowTitle() const from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1f
#2  0x7f6f35be39fe in QWidget::setWindowTitle(QString const&) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x2e
#3  /usr/lib/python3/dist-packages/PySide2/QtWidgets.cpython-37m-x86_64-linux-gnu.so(+0x62d21b) [0x7f6f111a021b]
#4  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyMethodDef_RawFastCallKeywords+0x2e8) [0x7f6f37197cb8]
#5  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyCFunction_FastCallKeywords+0x25) [0x7f6f37198a05]
#6  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalFrameDefault+0x7e90) [0x7f6f36f99cb0]
#7  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x794d3) [0x7f6f36f9c4d3]
#8  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyFunction_FastCallDict+0x2e2) [0x7f6f37198542]
#9  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyObject_Call_Prepend+0xcd) [0x7f6f3719b20d]
#10  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x215af5) [0x7f6f37138af5]
#11  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x21f072) [0x7f6f37142072]
#12  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyObject_FastCallKeywords+0xd3) [0x7f6f37199043]
#13  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalFrameDefault+0x52eb) [0x7f6f36f9710b]
#14  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalCodeWithName+0x996) [0x7f6f370c11e6]
#15  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyFunction_FastCallKeywords+0x93) [0x7f6f37198123]
#16  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalFrameDefault+0x5d1c) [0x7f6f36f97b3c]
#17  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalCodeWithName+0x996) [0x7f6f370c11e6]
#18  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyFunction_FastCallDict+0xae) [0x7f6f3719830e]
#19  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyObject_Call_Prepend+0xcd) [0x7f6f3719b20d]
#20  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x215af5) [0x7f6f37138af5]
#21  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x21f072) [0x7f6f37142072]
#22  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyObject_FastCallKeywords+0xd3) [0x7f6f37199043]
#23  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalFrameDefault+0x612e) [0x7f6f36f97f4e]
#24  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x794d3) [0x7f6f36f9c4d3]
#25  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyFunction_FastCallDict+0x2e2) [0x7f6f37198542]
#26  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyObject_Call_Prepend+0xcd) [0x7f6f3719b20d]
#27  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x215af5) [0x7f6f37138af5]
#28  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x21f072) [0x7f6f37142072]
#29  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyObject_FastCallKeywords+0xd3) [0x7f6f37199043]
#30  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalFrameDefault+0x52eb) [0x7f6f36f9710b]
#31  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalCodeWithName+0x996) [0x7f6f370c11e6]
#32  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyFunction_FastCallKeywords+0x93) [0x7f6f37198123]
#33  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalFrameDefault+0x9334) [0x7f6f36f9b154]
#34  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalCodeWithName+0x996) [0x7f6f370c11e6]
#35  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyFunction_FastCallDict+0xae) [0x7f6f3719830e]
#36  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyObject_Call_Prepend+0xcd) [0x7f6f3719b20d]
#37  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(PyObject_Call+0x75) [0x7f6f37198c75]
#38  0x7f6f3a63513d in Base::pyCall(_object*, _object*) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x28
#39  0x7f6f3acc6e0c in Gui::ViewProviderPythonFeatureImp::setEdit(int) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x2d2
#40  0x7f6effb6f18b in Gui::ViewProviderPythonFeatureT<PathGui::ViewProviderPath>::setEdit(int) from /home/ton/FREECAD/daily/freecad/Mod/Path/PathGui.so+0x27
#41  0x7f6f3ac71725 in Gui::ViewProvider::startEditing(int) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x2d
#42  0x7f6f3aca4ccf in Gui::ViewProviderDragger::startEditing(int) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x2f
#43  0x7f6f3a5ad64e in Gui::Document::setEdit(Gui::ViewProvider*, int, char const*) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x142a
#44  0x7f6effb6bed5 in PathGui::ViewProviderPath::doubleClicked() from /home/ton/FREECAD/daily/freecad/Mod/Path/PathGui.so+0x37
#45  0x7f6effb6f545 in Gui::ViewProviderPythonFeatureT<PathGui::ViewProviderPath>::doubleClicked() from /home/ton/FREECAD/daily/freecad/Mod/Path/PathGui.so+0x45
#46  0x7f6f3a94ae45 in Gui::TreeWidget::onDoubleClickItem(QTreeWidgetItem*) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x2fb
#47  0x7f6f3a94ab18 in Gui::TreeWidget::mouseDoubleClickEvent(QMouseEvent*) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x52
#48  0x7f6f35bf300f in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xd0f
#49  0x7f6f35c95cde in QFrame::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1e
#50  0x7f6f35e071eb in QAbstractItemView::viewportEvent(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1ab
[... cut ...]
#68  0x7f6f3523815b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13b
#69  0x7f6f35240132 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x92
#70  0x7f6f3a51fb40 in Gui::Application::runApplication() from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x5f2
#71  /home/ton/FREECAD/daily/freecad/bin/FreeCAD(+0x9168) [0x560533311168]
#72  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7f6f34aa409b]
#73  /home/ton/FREECAD/daily/freecad/bin/FreeCAD(+0x84da) [0x5605333104da]`

>

snelweg, 2021-09-10:
and another

> <Path.Area> PathUtils.py(314)|Area.cpp(1438): hit bottom -0,0,1e-06
Traceback (most recent call last):
  File "/home/ton/FREECAD/daily/freecad/Mod/Path/PathScripts/PathOpGui.py", line 104, in setEdit
    self.setupTaskPanel(TaskPanel(vobj.Object, self.deleteObjectsOnReject(), page, selection))
  File "/home/ton/FREECAD/daily/freecad/Mod/Path/PathScripts/PathOpGui.py", line 115, in setupTaskPanel
    panel.setupUi()
  File "/home/ton/FREECAD/daily/freecad/Mod/Path/PathScripts/PathOpGui.py", line 1210, in setupUi
    self.panelSetFields()
  File "/home/ton/FREECAD/daily/freecad/Mod/Path/PathScripts/PathOpGui.py", line 1182, in panelSetFields
    page.pageSetFields()
  File "/home/ton/FREECAD/daily/freecad/Mod/Path/PathScripts/PathOpGui.py", line 247, in pageSetFields
    self.setFields(self.obj)
  File "/home/ton/FREECAD/daily/freecad/Mod/Path/PathScripts/PathPocketBaseGui.py", line 144, in setFields
    self.setupCoolant(obj, self.form.coolantController)
  File "/home/ton/FREECAD/daily/freecad/Mod/Path/PathScripts/PathOpGui.py", line 404, in setupCoolant
    combo.blockSignals(True)
<class 'AttributeError'>: 'PySide2.QtWidgets.QTableWidgetItem' object has no attribute 'blockSignals'
Free Select
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libpthread.so.0(+0x12730) [0x7feb8a7f5730]
#1  0x7feb8a093627 in QLabel::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7
#2  /usr/lib/python3/dist-packages/PySide2/QtWidgets.cpython-37m-x86_64-linux-gnu.so(+0x425703) [0x7feb6137d703]
#3  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyMethodDef_RawFastCallKeywords+0x2e8) [0x7feb8b54acb8]
#4  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyMethodDescr_FastCallKeywords+0x4a) [0x7feb8b54b80a]
#5  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalFrameDefault+0x8f3c) [0x7feb8b34dd5c]
#6  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x794d3) [0x7feb8b34f4d3]
#7  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalFrameDefault+0x9334) [0x7feb8b34e154]
#8  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x794d3) [0x7feb8b34f4d3]
#9  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyFunction_FastCallDict+0x2e2) [0x7feb8b54b542]
#10  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyObject_Call_Prepend+0xcd) [0x7feb8b54e20d]
#11  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x215af5) [0x7feb8b4ebaf5]
#12  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x21f072) [0x7feb8b4f5072]
#13  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyObject_FastCallKeywords+0xd3) [0x7feb8b54c043]
#14  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalFrameDefault+0x52eb) [0x7feb8b34a10b]
#15  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(+0x794d3) [0x7feb8b34f4d3]
#16  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalFrameDefault+0x9334) [0x7feb8b34e154]
#17  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalCodeWithName+0x996) [0x7feb8b4741e6]
#18  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyFunction_FastCallDict+0xae) [0x7feb8b54b30e]
#19  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyObject_Call_Prepend+0xcd) [0x7feb8b54e20d]
#20  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(PyObject_Call+0x75) [0x7feb8b54bc75]
#21  0x7feb8e9e813d in Base::pyCall(_object*, _object*) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x28
#22  0x7feb8f079e0c in Gui::ViewProviderPythonFeatureImp::setEdit(int) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x2d2
#23  0x7feb53ea518b in Gui::ViewProviderPythonFeatureT<PathGui::ViewProviderPath>::setEdit(int) from /home/ton/FREECAD/daily/freecad/Mod/Path/PathGui.so+0x27
#24  0x7feb8f024725 in Gui::ViewProvider::startEditing(int) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x2d
#25  0x7feb8f057ccf in Gui::ViewProviderDragger::startEditing(int) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x2f
#26  0x7feb8e96064e in Gui::Document::setEdit(Gui::ViewProvider*, int, char const*) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x142a
#27  0x7feb53ea1ed5 in PathGui::ViewProviderPath::doubleClicked() from /home/ton/FREECAD/daily/freecad/Mod/Path/PathGui.so+0x37
#28  0x7feb53ea5545 in Gui::ViewProviderPythonFeatureT<PathGui::ViewProviderPath>::doubleClicked() from /home/ton/FREECAD/daily/freecad/Mod/Path/PathGui.so+0x45
#29  0x7feb8ecfde45 in Gui::TreeWidget::onDoubleClickItem(QTreeWidgetItem*) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x2fb
#30  0x7feb8ecfdb18 in Gui::TreeWidget::mouseDoubleClickEvent(QMouseEvent*) from /home/ton/FREECAD/daily/freecad/lib/libFreeCADGui.so+0x52
#31  0x7feb89fa600f in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xd0f
#32  0x7feb8a048cde in QFrame::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1e
#33  0x7feb8a1ba1eb in QAbstractItemView::viewportEvent(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1ab
#34  0x7feb8a2223eb in QTreeView::viewportEvent(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x3b
#35  0x7feb895ec19b in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x9b
#36  0x7feb89f674b1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x71
#37  0x7feb89f6ebb8 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x458
[... cut ...]
#55  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7feb88e5709b]
#56  /home/ton/FREECAD/daily/freecad/bin/FreeCAD(+0x84da) [0x55c80f3274da]

`OS: Debian GNU/Linux 10 (buster) (X-Cinnamon/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.28428 (Git)
Build type: Unknown
Branch: LinkDaily
Hash: 1e3a8d8c9e371d1f361cf2e5fe502b4a87bd6f86
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
`

realthunder, 2021-09-10:
Can you please describe your steps? Just double click to edit the path job? Anything special about the job?

luzpaz, 2022-01-18:
@snelweg bump

closed 2022-01-19 by snelweg