realthunder/FreeCAD #177: Navigation cude characters size.

onekk, 2021-09-10:
How to modify the Navigation Cube character size, they are almost unreadable, see image:

[image]

This is my version info

OS: Artix Linux (openbox)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3627 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: f63a5ea7b459872a3798027df0ff25e776524859
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: Italian/Italy (it_IT)

I have installed the stock FreeCAD appimage and everything is fine.

Regards

Carlo D.

bambukouk, 2021-09-15:
Have you tried cube menu addon?

wohltat, 2021-09-16:
I've tried the cube menu addon but i can't see any new settings. Of course i did a FC restart. Am i missing something here?
[image]

It seems as it should be possible to change the font size with the cube menu addon.
https://forum.freecadweb.org/viewtopic.php?f=34&t=43338&start=10#p379761

Nevertheless, i think the cube should work out of the box. The font size is even a bit small on my system but @onekk s cube font looks really tiny which i would consider a bug.

-------
```
OS: Linux Mint 20.1 (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```

realthunder, 2021-09-16:
Please try the latest 09.16 release.

[image]

wohltat, 2021-09-16:
Great. Works like in the video, so there is no need for the addon anymore.

bambukouk, 2021-09-16:
Proper job :+1:  
Works a treat - another addon gone ...

onekk, 2021-09-18:
I can confirm that the new version 20210916 (Stable) Improve the character size, and permit to change directly using the "small navigation cube", but the "predefined view icons" are disappeared.

There is a way to having them back?

Regards

Carlo D.

bambukouk, 2021-09-18:
Also, if one changes to either orthographic or perspective view than it seems impossible to get back to isometric view.
One can toggle between orto and perspective but no way to get back to iso?

bambukouk, 2021-09-18:
> .... but the "predefined view icons" are disappeared..

no, they just moved on - right click anywhere on the screen:

[image]

luzpaz, 2022-01-18:
Ticket looks like it can be closed.

closed 2022-01-18 by realthunder