realthunder/FreeCAD #178: Surfaces created with shape binders are not exported into a step file

wsteffe, 2021-09-14:
Hello,
 In the annexed file I used shape binders linked to a couple of faces in order to define input/output coaxial ports.
These faces are placed in separate bodies named WGP_1 and WGP_2 and I did expect they were exported into the step file. Unfortunately they didn't. The problem is related with the shape binders and not with the fact that the bodies contain surfaces (instead of solids). In fact there is another body (named SPLIT_1) which contains surfaces defined by a Part/Extrude and these are properly exported.

[CMP_KaCh3_SFILT_6_4.zip](https://github.com/realthunder/FreeCAD/files/7162827/CMP_KaCh3_SFILT_6_4.zip)

wsteffe, 2021-09-16:
Hello RT, if for some reason it is not possible to export surfaces defined in shape binders may you please suggest me another way to put in a Body, a copy of a face of a solid defined in a different Body in such a way that this copy is exported in a step file (under a product name defined by the name of the first body).
I have already tried with Draft Facebinder, but if I use this feature after having activated a body, this body becomes totally screwed. I think that that this is due to a strong incompatibility between Draft WB and your Part Design WB.
I have also tried with Part Offset but this feature does not work when the offset distance is 0. And I think that this is another bug. In other CAD systems (Catia and ZW3D) offset (or copy/move) works also with a 0 distance.

wohltat, 2021-09-16:
My FreeCAD crashes if i try to open the `CMP_KaCh3_SFILT_6_4.FCStd` file.
```
<Import> ImportOCAF2.cpp(659): free shape count 1
/tmp/.mount_FreeCAQot4au/AppRun: line 46: 38364 Segmentation fault      (core dumped) "$PREFIX/bin/$name" "$@"
```

I've tried to export a shapebinder of a surface of a cone to .stp file and it did work.

@wsteffe Which Version are you using? Also it is a good habit to always add the FreeCAD version string when filing an issue as well as using a minimal example.

------
```
OS: Linux Mint 20.1 (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```

wsteffe, 2021-09-16:
Hello Wholtat, thanks for trying it. I do not know why it makes your FreeCAD crash.
The document was created with FC LinkDaily built from sources synced from this repository a few days ago.  Following the version string printed from my running FreeCAD.

OS: Ubuntu 20.04.3 LTS (GNOME-Flashback:GNOME/gnome-flashback-metacity)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.28450 (Git)
Build type: Unknown
Branch: LinkDaily
Hash: 24898975854047b8b49adbfc23ceea17a4c77f4d
Python version: 3.8.10
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.5.0
Locale: English/United States (en_US)

realthunder, 2021-09-16:
I will enable `SubShapeBinder` export in the future. In the meantime, you can use `Boolean` feature for exporting. Assuming you've already created the `SubShapeBinder` in a body that is bound to a face. Select the binder, and then create a `Boolean`. In its edit panel, check `New shape` option. This way the surface becomes a standalone shape without causing error trying to fuse to others. Now, right click the `Boolean` and select `Toggle export`. Note that enable export this way will disable the original 'export tip` function of the body. You'll need to explicitly select which shape(s) to export from the body.

wsteffe, 2021-09-16:
Hello, RT.
Thanks for the hint. It works but I would like to leave this issue open because, as you said, you are going to enable SubShapeBinder export in the future.

I would also say that I have just observed a strange new behaviour of my FC.
When I toggle an active body it becomes bold as expected, but the name of the previously active body also remains bold. So I may have several Bodies that seem active and I can't understand which is the real one. I think it didn't happen before my last sync with (yesterday). The updated document in which I have observed this problem is here annexed but I am not sure that the problem is related with this particular document. [CMP_KaCh3_SFILT_6_4.zip](https://github.com/realthunder/FreeCAD/files/7177901/CMP_KaCh3_SFILT_6_4.zip)

commit e662af10 (2021-09-17): Part: enable 'Toggle export' of SubShapeBinder in body

Closes #178
commit d271cbf1 (2021-09-21): Part: enable 'Toggle export' of SubShapeBinder in body

Closes #178
closed 2021-09-17 by realthunder