realthunder/FreeCAD #181: Task as a separate pop up window

wsteffe, 2021-09-18:
Several tasks require the selection of more than one  geometrical element to fill different entries.
In example to make a sweep you need to select a base profile and a path ...
Generally I prefer to make these selections in the model tree but I do not like to switch every time fort and back between Task and Model Tabs. In my opinion it would be much better if the active task were a separate pop up window and the Model were always visible and selectable. The Task window should disappear when the task is closed.  It should not be a persistent element of the document structure.

realthunder, 2021-09-18:
Change `Preferences -> General -> Tree view mode` to `TreeView and PropertyView` will get what you want. Or, if you are okay with overlay interface, try `Tools -> Preset configurations`, either `Overlay dark` or `Overlay light`. The task panel becomes transparent and mouse click through if no task dialog.

wsteffe, 2021-09-18:
Hello RT, thanks for clarification.
In fact the "TreeView and PropertyView" with Task Panel toggled to Overlay is a very good setting for me.
I have always the possibility to view and select the model elements and the Task Panel doesn't waste space when not active.

closed 2021-09-21 by realthunder