realthunder/FreeCAD #187: python console false typing behavior when autocomplete suggestions are shown

wohltat, 2021-09-23:
If i type in the python console and the autosuggestions are shown, then some keys are not typed. This seems to be connected to the shortcut keybindings which should not be accessible in the console. For example if i press 'r' + 'r' i would expect it to type two 'r's but instead the r-pie-menu is triggered as if i would focus the main window / working area. 
This happens with other shortcut keys as well. 

Maybe this is connected to the fix for #167 since a can't remember seeing this before.

[image]

-------------

OS: Linux Mint 20.1 (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.916.24301 +4152 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 7fd36f96f4db17db4f8d058eab22588cca8f9d5c
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

realthunder, 2021-09-24:
Please try the latest 09.22 release. There is a fix for this problem.

wohltat, 2021-09-24:
Yep it's fixed now, thanks.

I've activated the github Watch for new releases now, so i don't need to file bugs anymore one day after the fix came out. ;)

closed 2021-09-24 by wohltat