realthunder/FreeCAD #190: Compile fails with latest daily

snelweg, 2021-09-24:
`cd FREECAD/daily/freecad-source
ton@DESKTOP-9FIBCI1:~/FREECAD/daily/freecad-source$ git pull
remote: Enumerating objects: 15, done.
remote: Counting objects: 100% (15/15), done.
remote: Compressing objects: 100% (1/1), done.
remote: Total 8 (delta 7), reused 8 (delta 7), pack-reused 0
Unpacking objects: 100% (8/8), done.
From github.com:realthunder/FreeCAD
   3de879c7a2..43b2b5639b  LinkDaily  -> origin/LinkDaily
Updating 3de879c7a2..43b2b5639b
Fast-forward
 src/Mod/Part/App/ShapeUpgrade/UnifySameDomainPyImp.cpp | 1 +
 1 file changed, 1 insertion(+)
ton@DESKTOP-9FIBCI1:~/FREECAD/daily/freecad-source$ cd ../freecad
ton@DESKTOP-9FIBCI1:~/FREECAD/daily/freecad$ cmake   ../freecad-source -DBUILD_FEM=OFF -DBUILD_QT5=ON -DPYTHON_EXECUTABLE=/usr/bin/python3.7 
-- Install configuration: ""
-- Compiler: GNU, version: 8.3.0
-- prefix: /usr/local
-- bindir: bin
-- datadir: share
-- docdir: share/doc/FreeCAD
-- includedir: include
-- libdir: lib
-- cmake: 3.13.4
-- BUILD_MESH_PART requires BUILD_SMESH to be ON, but it is ""
-- BUILD_OPENSCAD requires BUILD_MESH_PART to be ON, but it is "OFF"
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   filesystem
--   program_options
--   regex
--   system
--   thread
--   chrono
--   date_time
--   atomic
-- PyCXX found:
--   Headers:  /home/ton/FREECAD/daily/freecad-source/src
--   Sources:  /home/ton/FREECAD/daily/freecad-source/src/CXX
--   Version:  6.2.8
-- -- Found OCE/OpenCASCADE version: 7.3.0
-- -- OCE/OpenCASCADE include directory: /usr/include/opencascade
-- -- OCE/OpenCASCADE shared libraries directory: /usr/lib/x86_64-linux-gnu
-- Performing Test _flag_found
-- Performing Test _flag_found - Success
-- Checking for module 'Coin'
--   Package 'Coin', required by 'virtual:world', not found
-- Coin3D libraries found
-- Shiboken2Config: Using default python: -python2.7.x86_64-linux-gnu
-- libshiboken built for Release
-- PYTHON_CONFIG_SUFFIX: .cpython-37m-x86_64-linux-gnu
-- libshiboken built for Release
=======================
PySide2Tools not found.
=======================

-- ===============================================
PySide2 Python module found at /usr/lib/python3/dist-packages/PySide2.
===============================================

-- -- matplotlib-3.0.2 has been found.
-- Platform is 64-bit, set -D_OCC64
-- Performing Test _flag_found
-- Performing Test _flag_found - Success
-- Performing Test _flag_found
-- Performing Test _flag_found - Success
-- setting gcc options: -Wall -Werror -Wno-deprecated -pedantic-errors
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   python37
[... 78 lines cut ...]
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `Qt::hex(QTextStream&)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QString::split(QChar, QFlags<Qt::SplitBehaviorFlags>, Qt::CaseSensitivity) const@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `operator delete(void*, unsigned long)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QMetaCallEvent::QMetaCallEvent(unsigned short, unsigned short, void (*)(QObject*, QMetaObject::Call, int, void**), QObject const*, int, int)@Qt_5_PRIVATE_API'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QRecursiveMutex::~QRecursiveMutex()@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QJsonValue::QJsonValue(QJsonValue&&)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QMetaType::name() const@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QJsonArray::QJsonArray(QJsonArray&&)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QtPrivate::argToQString(QLatin1String, unsigned long, QtPrivate::ArgBase const**)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QThread::wait(QDeadlineTimer)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QJsonObject::QJsonObject(QJsonObject&&)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QJsonValue::swap(QJsonValue&)@Qt_5'
collect2: error: ld returned 1 exit status
make[2]: *** [src/Main/CMakeFiles/FreeCADMain.dir/build.make:112: bin/FreeCAD] Error 1
make[1]: *** [CMakeFiles/Makefile2:562: src/Main/CMakeFiles/FreeCADMain.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 80%] Linking CXX shared library ../../../../Mod/Part/PartGui.so
[ 81%] Built target MeshGui
[ 81%] Built target WebGui
[ 81%] Built target RaytracingGui
[ 82%] Built target RobotGui
[ 82%] Built target DrawingGui
[ 85%] Built target PartGui
make: *** [Makefile:130: all] Error 2
`

snelweg, 2021-09-24:
Must be my mistake, even when I rollback FreeCAD will still not compile.

realthunder, 2021-09-24:
Have you installed any library update? It looks like something wrong with the WebGui module. Maybe try to rebuild the Web module by removing `build/Mod/Web` and `build/src/Mod/Web` directory.

snelweg, 2021-09-24:
> Have you installed any library update? It looks like something wrong with the WebGui module. Maybe try to rebuild the Web module by removing `build/Mod/Web` and `build/src/Mod/Web` directory.

I removed the whole freecad build directory and started from scratch, sorry to say it still doesn't compile. I must have broke something while installing LinuxCNC on this Debian Buster machine.

```
Building CXX object src/Mod/Test/Gui/CMakeFiles/QtUnitGui.dir/AppTestGui.cpp.o
[ 83%] Generating moc_DlgFilletEdges.cpp
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QNetworkRequest::QNetworkRequest()@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QStringRef::split(QChar, QFlags<Qt::SplitBehaviorFlags>, Qt::CaseSensitivity) const@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QWaitCondition::wait(QMutex*, QDeadlineTimer)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `Qt::dec(QTextStream&)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QRecursiveMutex::QRecursiveMutex()@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QString::splitRef(QChar, QFlags<Qt::SplitBehaviorFlags>, Qt::CaseSensitivity) const@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QMetaProperty::isRequired() const@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `operator delete[](void*, unsigned long)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `qt_version_tag@Qt_5.15'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QtPrivate::argToQString(QStringView, unsigned long, QtPrivate::ArgBase const**)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `Qt::right(QTextStream&)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QDate::startOfDay(Qt::TimeSpec, int) const@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `Qt::hex(QTextStream&)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QString::split(QChar, QFlags<Qt::SplitBehaviorFlags>, Qt::CaseSensitivity) const@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `operator delete(void*, unsigned long)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QMetaCallEvent::QMetaCallEvent(unsigned short, unsigned short, void (*)(QObject*, QMetaObject::Call, int, void**), QObject const*, int, int)@Qt_5_PRIVATE_API'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QRecursiveMutex::~QRecursiveMutex()@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QJsonValue::QJsonValue(QJsonValue&&)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QMetaType::name() const@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QJsonArray::QJsonArray(QJsonArray&&)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QtPrivate::argToQString(QLatin1String, unsigned long, QtPrivate::ArgBase const**)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QThread::wait(QDeadlineTimer)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QJsonObject::QJsonObject(QJsonObject&&)@Qt_5'
/usr/bin/ld: /home/ton/Qt2/5.15.2/gcc_64/lib//libQt5Qml.so.5: undefined reference to `QJsonValue::swap(QJsonValue&)@Qt_5'
collect2: error: ld returned 1 exit status
make[2]: *** [src/Main/CMakeFiles/FreeCADMain.dir/build.make:112: bin/FreeCAD] Error 1
make[1]: *** [CMakeFiles/Makefile2:562: src/Main/CMakeFiles/FreeCADMain.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 83%] Generating moc_DlgPartBoxImp.cpp
[ 83%] Generating moc_RemeshGmsh.cpp
[ 83%] Generating moc_DlgPartCylinderImp.cpp
[ 83%] Building CXX object src/Mod/Image/Gui/CMakeFiles/ImageGui.dir/AppImageGuiPy.cpp.o
[ 83%] Building CXX object src/Mod/Test/Gui/CMakeFiles/QtUnitGui.dir/PreCompiled.cpp.o
[ 83%] Generating moc_DlgPartImportIgesImp.cpp
[ 83%] Linking CXX shared library ../../../../Mod/Test/QtUnitGui.so
[ 83%] Generating moc_DlgPartImportStepImp.cpp
[ 84%] Building CXX object src/Mod/Image/Gui/CMakeFiles/ImageGui.dir/Command.cpp.o
[ 84%] Generating moc_DlgPrimitives.cpp
[ 84%] Generating moc_SegmentationBestFit.cpp
[ 84%] Built target QtUnitGui
[ 84%] Generating moc_Selection.cpp
[ 84%] Generating moc_MeshEditor.cpp
[ 84%] Generating moc_DlgRevolution.cpp
[ 84%] Building CXX object src/Mod/Image/Gui/CMakeFiles/ImageGui.dir/ImageOrientationDialog.cpp.o
[ 84%] Building CXX object src/Mod/Image/Gui/CMakeFiles/ImageGui.dir/ViewProviderImagePlane.cpp.o
Scanning dependencies of target MeshGui
[ 84%] Generating moc_DlgSettings3DViewPartImp.cpp
[ 84%] Building CXX object src/Mod/Points/Gui/CMakeFiles/PointsGui.dir/Command.cpp.o
[ 84%] Generating moc_DlgSettingsGeneral.cpp
[ 84%] Generating moc_DlgSettingsObjectColor.cpp
[ 84%] Generating moc_DlgProjectionOnSurface.cpp
[... cut ...]
[ 87%] Building CXX object src/Mod/Part/Gui/CMakeFiles/PartGui.dir/TaskCheckGeometry.cpp.o
[ 87%] Building CXX object src/Mod/Part/Gui/CMakeFiles/PartGui.dir/TaskAttacher.cpp.o
[ 87%] Building CXX object src/Mod/Part/Gui/CMakeFiles/PartGui.dir/ViewProviderPartExtPyImp.cpp.o
[ 87%] Building CXX object src/Mod/Part/Gui/CMakeFiles/PartGui.dir/PartParams.cpp.o
[ 87%] Linking CXX shared library ../../../../Mod/Part/PartGui.so
[ 87%] Built target PartGui
make: *** [Makefile:130: all] Error 2

real    23m4.528s
user    126m13.911s
sys     13m29.031s
```

realthunder, 2021-09-24:
Can you please post your `CMakeCache.txt` file here? For some reason you are linking to Qt 5.15.2, but according to your previous cmake output. You only have Qt 5.11.3 installed.

snelweg, 2021-09-24:
> Can you please post your `CMakeCache.txt` file here? For some reason you are linking to Qt 5.15.2, but according to your previous cmake output. You only have Qt 5.11.3 installed.

I appreciate your effort, I am very sorry to take away your time in solving my stupid mistakes.

[CMakeCache.txt](https://github.com/realthunder/FreeCAD/files/7224848/CMakeCache.txt)

snelweg, 2021-09-24:
I do have several Qt5 versions installed on my machine. Maybe I made a mistake there?

```
DESKTOP-9FIBCI1:~/Qt2$ ls -la
total 30028
drwxr-xr-x 13 ton ton     4096 Jul 18 19:08 .
drwxr-xr-x 57 ton ton     4096 Sep 24 10:40 ..
drwxrwxr-x  6 ton ton     4096 Jul 18 18:58 5.12.8
drwxrwxr-x  6 ton ton     4096 Jul 18 18:58 5.12.9
drwxrwxr-x  4 ton ton     4096 Jul 18 18:58 5.15.0
drwxrwxr-x  4 ton ton     4096 Jul 18 18:58 5.15.1
drwxrwxr-x  6 ton ton     4096 Jul 18 18:59 5.15.2
-rw-r--r--  1 ton ton    67355 Jul 18 19:01 components.xml
drwxrwxr-x  2 ton ton     4096 Jul 18 18:58 dist
drwxrwxr-x  7 ton ton     4096 Jul 18 18:58 Docs
drwxrwxr-x  7 ton ton     4096 Jul 18 19:00 Examples
-rw-r--r--  1 ton ton   568990 Aug 25 10:16 InstallationLog.txt
-rw-r--r--  1 ton ton       48 Jul 18 19:01 installer.dat
drwxr-xr-x 96 ton ton     4096 Jul 18 19:01 installerResources
drwxr-xr-x  2 ton ton     4096 Jul 18 19:00 Licenses
-rwxr-xr-x  1 ton ton 29266600 Jul 18 19:01 MaintenanceTool
-rw-r--r--  1 ton ton   755738 Jul 18 19:01 MaintenanceTool.dat
-rw-r--r--  1 ton ton    20115 Jul 18 19:01 MaintenanceTool.ini
-rw-r--r--  1 ton ton      362 Jul 18 19:01 network.xml
drwxrwxr-x 10 ton ton     4096 Jul 18 19:00 Tools
```

realthunder, 2021-09-24:
I suspect you have upgrade the Qt5WebKit component causing version mismatch. Can you please double check the Qt5WebKit version on your computer? What distro are you using?

snelweg, 2021-09-24:
> 
> 
> I suspect you have upgrade the Qt5WebKit component causing version mismatch. Can you please double check the Qt5WebKit version on your computer? What distro are you using?

I will when I am at the machine. Thanks!

snelweg, 2021-09-25:
> I suspect you have upgrade the Qt5WebKit component causing version mismatch. Can you please double check the Qt5WebKit version on your computer? What distro are you using?

Using:
`Linux DESKTOP-9FIBCI1 4.19.0-17-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.194-3 (2021-07-18) x86_64 GNU/Linux
`

Qt5WebKit version:
`/usr/lib/x86_64-linux-gnu/libQt5WebKit.so.5.212.0
`

Is that the information you wanted?

snelweg, 2021-09-25:
Good news!

I did a 
`export LD_LIBRARY_PATH=/home/ton/Qt2/5.12.9/gcc_64/lib/`
and then the regular
`ton@DESKTOP-9FIBCI1:~/FREECAD/daily/freecad$ time cmake --build /home/ton/FREECAD/daily/freecad -j 7
and it compiled and I have a working version of your daily release. Very happy, thanks for pointing out what I did wrong,
`

snelweg, 2021-09-25:
oops spoke to soon,
FC compiles without errors but doesn't work at all.

```
~/FREECAD/daily/freecad/bin/FreeCAD
FreeCAD 0.19, Libs: 0.19R28493 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2021
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

PySide2/__init__.py: Unable to import shiboken2 from /home/ton/FREECAD/daily/freecad/Mod/Measure, /home/ton/FREECAD/daily/freecad/Mod/Surface, /home/ton/FREECAD/daily/freecad/Mod/Material, /home/ton/FREECAD/daily/freecad/Mod/Spreadsheet, /home/ton/FREECAD/daily/freecad/Mod/Web, /home/ton/FREECAD/daily/freecad/Mod/Points, /home/ton/FREECAD/daily/freecad/Mod/Idf, /home/ton/FREECAD/daily/freecad/Mod/Part, /home/ton/FREECAD/daily/freecad/Mod/Mesh, /home/ton/FREECAD/daily/freecad/Mod/Import, /home/ton/FREECAD/daily/freecad/Mod/Start, /home/ton/FREECAD/daily/freecad/Mod/AddonManager, /home/ton/FREECAD/daily/freecad/Mod/Path, /home/ton/FREECAD/daily/freecad/Mod/Sketcher, /home/ton/FREECAD/daily/freecad/Mod/Test, /home/ton/FREECAD/daily/freecad/Mod/Raytracing, /home/ton/FREECAD/daily/freecad/Mod/TechDraw, /home/ton/FREECAD/daily/freecad/Mod/Drawing, /home/ton/FREECAD/daily/freecad/Mod/Show, /home/ton/FREECAD/daily/freecad/Mod/Robot, /home/ton/FREECAD/daily/freecad/Mod/Arch, /home/ton/FREECAD/daily/freecad/Mod/Image, /home/ton/FREECAD/daily/freecad/Mod/Draft, /home/ton/FREECAD/daily/freecad/Mod/Inspection, /home/ton/FREECAD/daily/freecad/Mod/Tux, /home/ton/FREECAD/daily/freecad/Mod/PartDesign, /home/ton/FREECAD/daily/freecad/Mod/ReverseEngineering, /home/ton/FREECAD/daily/freecad/Mod, /home/ton/FREECAD/daily/freecad/lib, /home/ton/FREECAD/daily/freecad/Ext, /home/ton/FREECAD/daily/freecad/bin, /usr/lib/python37.zip, /usr/lib/python3.7, /usr/lib/python3.7/lib-dynload, /home/ton/.local/lib/python3.7/site-packages, /usr/local/lib/python3.7/dist-packages, /usr/lib/python3/dist-packages, /home/ton/.FreeCAD/Macro, /home/ton/FREECAD/daily/freecad/Macro
During initialization the error "name '_init_pyside_extension' is not defined" occurred in /home/ton/FREECAD/daily/freecad/Mod/Tux/InitGui.py
Please look into the log file for further information
PySide2/__init__.py: Unable to import shiboken2 from /home/ton/FREECAD/daily/freecad/Mod/Measure, /home/ton/FREECAD/daily/freecad/Mod/Surface, /home/ton/FREECAD/daily/freecad/Mod/Material, /home/ton/FREECAD/daily/freecad/Mod/Spreadsheet, /home/ton/FREECAD/daily/freecad/Mod/Web, /home/ton/FREECAD/daily/freecad/Mod/Points, /home/ton/FREECAD/daily/freecad/Mod/Idf, /home/ton/FREECAD/daily/freecad/Mod/Part, /home/ton/FREECAD/daily/freecad/Mod/Mesh, /home/ton/FREECAD/daily/freecad/Mod/Import, /home/ton/FREECAD/daily/freecad/Mod/Start, /home/ton/FREECAD/daily/freecad/Mod/AddonManager, /home/ton/FREECAD/daily/freecad/Mod/Path, /home/ton/FREECAD/daily/freecad/Mod/Sketcher, /home/ton/FREECAD/daily/freecad/Mod/Test, /home/ton/FREECAD/daily/freecad/Mod/Raytracing, /home/ton/FREECAD/daily/freecad/Mod/TechDraw, /home/ton/FREECAD/daily/freecad/Mod/Drawing, /home/ton/FREECAD/daily/freecad/Mod/Show, /home/ton/FREECAD/daily/freecad/Mod/Robot, /home/ton/FREECAD/daily/freecad/Mod/Arch, /home/ton/FREECAD/daily/freecad/Mod/Image, /home/ton/FREECAD/daily/freecad/Mod/Draft, /home/ton/FREECAD/daily/freecad/Mod/Inspection, /home/ton/FREECAD/daily/freecad/Mod/Tux, /home/ton/FREECAD/daily/freecad/Mod/PartDesign, /home/ton/FREECAD/daily/freecad/Mod/ReverseEngineering, /home/ton/FREECAD/daily/freecad/Mod, /home/ton/FREECAD/daily/freecad/lib, /home/ton/FREECAD/daily/freecad/Ext, /home/ton/FREECAD/daily/freecad/bin, /usr/lib/python37.zip, /usr/lib/python3.7, /usr/lib/python3.7/lib-dynload, /home/ton/.local/lib/python3.7/site-packages, /usr/local/lib/python3.7/dist-packages, /usr/lib/python3/dist-packages, /home/ton/.FreeCAD/Macro, /home/ton/FREECAD/daily/freecad/Macro
During initialization the error "name '_init_pyside_extension' is not defined" occurred in /home/ton/FREECAD/daily/freecad/Mod/Draft/InitGui.py
Please look into the log file for further information
PySide2/__init__.py: Unable to import shiboken2 from /home/ton/FREECAD/daily/freecad/Mod/Measure, /home/ton/FREECAD/daily/freecad/Mod/Surface, /home/ton/FREECAD/daily/freecad/Mod/Material, /home/ton/FREECAD/daily/freecad/Mod/Spreadsheet, /home/ton/FREECAD/daily/freecad/Mod/Web, /home/ton/FREECAD/daily/freecad/Mod/Points, /home/ton/FREECAD/daily/freecad/Mod/Idf, /home/ton/FREECAD/daily/freecad/Mod/Part, /home/ton/FREECAD/daily/freecad/Mod/Mesh, /home/ton/FREECAD/daily/freecad/Mod/Import, /home/ton/FREECAD/daily/freecad/Mod/Start, /home/ton/FREECAD/daily/freecad/Mod/AddonManager, /home/ton/FREECAD/daily/freecad/Mod/Path, /home/ton/FREECAD/daily/freecad/Mod/Sketcher, /home/ton/FREECAD/daily/freecad/Mod/Test, /home/ton/FREECAD/daily/freecad/Mod/Raytracing, /home/ton/FREECAD/daily/freecad/Mod/TechDraw, /home/ton/FREECAD/daily/freecad/Mod/Drawing, /home/ton/FREECAD/daily/freecad/Mod/Show, /home/ton/FREECAD/daily/freecad/Mod/Robot, /home/ton/FREECAD/daily/freecad/Mod/Arch, /home/ton/FREECAD/daily/freecad/Mod/Image, /home/ton/FREECAD/daily/freecad/Mod/Draft, /home/ton/FREECAD/daily/f [... cut ...] /FREECAD/daily/freecad/Mod/ReverseEngineering, /home/ton/FREECAD/daily/freecad/Mod, /home/ton/FREECAD/daily/freecad/lib, /home/ton/FREECAD/daily/freecad/Ext, /home/ton/FREECAD/daily/freecad/bin, /usr/lib/python37.zip, /usr/lib/python3.7, /usr/lib/python3.7/lib-dynload, /home/ton/.local/lib/python3.7/site-packages, /usr/local/lib/python3.7/dist-packages, /usr/lib/python3/dist-packages, /home/ton/.FreeCAD/Macro, /home/ton/FREECAD/daily/freecad/Macro
name '_init_pyside_extension' is not defined
```

snelweg, 2021-09-26:
I gave up and reinstalled my machine with a fresh Debian 11 version. At least I can use the daily appimages again and get to work.
Thanks for your help!

closed 2021-09-26 by snelweg

snelweg, 2021-09-26:
Good decision to upgrade Debian. Everything works straight out the box.
Compile time just 32 minutes now!

`Linux DESKTOP-9FIBCI1 5.10.0-8-amd64 #1 SMP Debian 5.10.46-5 (2021-09-23) x86_64 GNU/Linux`

```
[100%] Linking CXX shared library ../../../../Mod/PartDesign/PartDesignGui.so
[100%] Built target PartDesignGui

real	32m12.386s
user	190m9.789s
sys	14m24.954s

ton@DESKTOP-9FIBCI1:~/FREECAD/daily/freecad$ bin/FreeCAD
FreeCAD 0.19, Libs: 0.19R28504 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2021
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##
```