realthunder/FreeCAD #192: Debugging a FC document

wsteffe, 2021-09-27:
It was already said in other posts that an FC document doesn't have a linear history which you may explore going forward and backward through the movement of a line in the model tree.
This possibility is offered by other CADs (in example in ZW3D) but not in FC. 
In FC there is just a DAG which may be considered a more essential data structure because it gives to the computational code the possibility to follow different paths (all those that are compatible with DAG partial ordering). This may be good for computational efficiency (exploiting parallelism) but, in my opinion, there is a price to pay which is given by a much more difficult (if not impossible)  debugging.
Among all the possible computational paths there is a special one: that one actually followed in the first definition of the FC document. 
In my opinion this path (the real history of the document) plays an important role because, when I reopen a feature in the model tree (by double clicking on it) I would expect to find the same situation encountered when that feature was firstly created. At that time some features were already created and other were not. If it were possible to recover that situation, it would be much easier to fix a broken feature.

wsteffe, 2021-09-28:
RT, I would like to better clarify what I would like to have:

I would like to have an other view, named History, beside the model tree. This view should hold an ordered list of all features of all bodies. Initially the order should be defined by the creation time but the user should have the capability of reordering it. The reordering should be allowed only when it doesn't break the DAG partial ordering.

There should be also a movable line which, normally, is positioned at the and of the History. For debugging purpose the line may be moved at a different position. After the movement the CAD geometry should be cleared and only the part which comes before the line should be regenerated. The user would then be able to see the situation in which the next feature (the next one after the line) was created.

realthunder, 2021-09-28:
For FreeCAD, linear history view only makes sense for PartDesign features. And it does provide a similar view to other CAD software. The history can be viewed by simply toggle the visibility of each feature. And you can reorder them by drag and drop. In my branch, you can use the `New shape` option to create a branch inside the body which form its own linear feature. You can right click body the use the `Enable auto solid group` to organize each branch into its own group.

wsteffe, 2021-09-28:
> linear history view only makes sense for PartDesign features
I agree.

> The history can be viewed by simply toggle the visibility of each feature
That means that hidden is equivalent to not generated at all. It is interesting. The last part of my recipe "clear all and regenerate only what comes before the line" can then be replaced with "hide all and unhide only what comes before the line".

But, in my opinion, the History widget is still a nice thing to have. It may be very useful when there are cross references among different bodies.

Let us consider following design flow:
1) Make a new Body1 and a new Box1 inside of Body1.
2) Make a new Body2 and a new Sketch1 in Body2.
 Sketch1 is based on a face of Box1 and it is using Edge1 of that face as a reference.
3) Make a new Pad1 (still in Body2) from Sketch1
3) Activate Body1 and make a Fillet1 on Edge1.

Now I want to change (reopen) Sketch1 and I want to do it looking at the same situation I saw when I firstly created Sketch1. It is clear that I must hide not only the Pad1 (the subsequent feature in Body2) but also the Fillet1 which belongs to Body1 and was done after Sketch1.

With this simple structure it is easy to remember the history and to decide what must be hidden and not.
But in a more complex structure with many bodies and many cross references it may be very difficult to remember which features in which bodies were involved in the creation of a given feature.
And the difficulty to remember increases when you came back to that complex document after a long time.
In my opinion having a registered history, which my be executed step by step, helps a long to remember how the document was created and to understand the relationships among the features of its bodies.

wsteffe, 2021-09-28:
Helle RT, I would like to suggest also another different way (which I think you will prefer over history) to help with the debugging: A new command aimed to hide all feature on which the selected feature does not depend on.
In my previous example if this commend were executed after having selected Sketch1, it would hide Pad1 (because it follows Sketch1 in Body2) and Fillet1 because it follows Box1 (the last referred object of Body1).

As a general rule a future should be hidden if it is not referred directly or indirectly by the selected feature.
If there were another body containing a feature referred by Box1 (directly referred by Sketch1) that feature (indirectly referred by Sketch1) should also be made visible.  The command would automatically recreate a minimal set of past events (minimal history) which are necessary before the creation of Sketch1.

wsteffe, 2021-12-16:
I am closing this issue to reopen it under a slightly different perspective

closed 2021-12-16 by wsteffe