realthunder/FreeCAD #193: Linear Feature doesn't include its base feature in translations/preview

noob8650, 2021-10-03:
Hello dear realthunder,

I am having issues with the linear pattern feature.
How to recreate:
Have a cuboid, make a pocket, make a linear pattern of this pocket.
The created pockets will have one less than the selected "Occurrences" from the parameters panel, trying to offset the linear pattern will likewise only shift the subsequent pockets, not the parent pocket as previously.

An image is worth a thousand words:
https://i.postimg.cc/fzk33nQw/bug-linear-pattern.png
I am note sure if this is a bug or a feature, but it seems weird, that it doesn't work as previously.

OS: Ubuntu 21.04 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.922.24301 +3888 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 399863d7e37b354fea6687767605e661beace504
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

realthunder, 2021-10-04:
Please try the `Daily` branch release. Just look for the `Daily` keyword in the release assets. There is some changes of behavior in pattern feature that may fit your needs. The change is somewhat substantial, which is why it has not been ported to the `Stable` release yet.

noob8650, 2021-10-04:
I tried and downloaded the "Daily" Appimage
                                                               '-->https://github.com/realthunder/FreeCAD_assembly3/releases
but the "issue" seems to remain. Was this as intended?
Or is the first feature supposed to be part of the linear pattern?
Any issue aside, I want to really thank you, this is absolutely awesome and so are you! :)
It's really amazing to see what can be done (if only by a few heh).[image]
 

Version details:
OS: Ubuntu 21.04 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.922.24301 +4188 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 3de879c7a2ba7111cc769d98636eb6d43189459d
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

noob8650, 2021-10-04:
I previously used the same .fcstd file I had worked in in "Stable" when I tested if it would be 11 Occurrences instead of 10 in "Daily".
Creating a new file from scratch in "Daily" resulted in proper behavior and as visible in the image attached it works as expected.

[image]

Version: 2021.922.24301 +4188 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 3de879c7a2ba7111cc769d98636eb6d43189459d

May bad, but thanks for giving me the hint to use "Daily", I've been stuck on this until you answered.
Is it proper to close the issue now?

noob8650, 2021-10-04:
I spoke too soon, looks like it has the same odd behavior again
when I use multiple base features (1,2,3) for my linear pattern at the end.
Perhaps it has something to do with having the Local_CS in there; or the multiple features?
I'll attach the file.

[image]

Hash: 3de879c7a2ba7111cc769d98636eb6d43189459d

[heatsink_45x45x10_try6.tar.gz](https://github.com/realthunder/FreeCAD/files/7280531/heatsink_45x45x10_try6.tar.gz)

realthunder, 2021-10-06:
This is not really a mistake, but a design choice. It is also a work in progress, so there is surely room for improvement. The logic to determine whether to transform the first instance of the pattern is a bit complex, because moving the first instance, i.e. the original patterning feature, is essentially rewriting the history, which is not always possible. For your last posted model, it is disabled because you are patterning another pattern feature, which can potentially have its own complication of the 'first instance' problem. In other words, we could potentially facing a "rewriting an already rewritten history" problem.

I have changed the behavior with the above commit to disregard whether the patterning feature is itself a pattern, and let user deal with the potential complication. The criteria has been changed to enable 'first instance' transformation if all patterning features belonging to the same history group are consecutive immediate history features of this pattern feature. BTW, as a user, you are advised to not apply transformation offset to a pattern of pattern. You can always transform the first pattern to the desired the location, and apply the second pattern without offset.

As to why you can't get the new behavior in `Daily` with your old file, there is a hidden property `_Version` in the pattern feature to make it behave the same in different builds. You can reveal it by select the pattern feature, right click in the property editor, and select `Show all`. This also applies to the new behavior here. If you want to get the new behavior in future release without rebuild your model, you'll need to manually change `_Version` to 3.

noob8650, 2021-10-06:
Tanks very much for your detailed response, that's really helpful to know and I'll make sure to spread that detail.
The detail about the version is really good to know, thanks for taking the time to explain it to me.
I'll try harder to make sure it's not a user error next time. :sweat_smile: 
Have a nice day!

commit d588fae5 (2021-10-06): PartDesign: change pattern transform first pattern instance behavior

Apply transformation offset to the first instance of the pattern if all
the patterning features of the same sibling group of this pattern
feature are consecutive immediate history. See added code comments for
more details.

Fixes #193