realthunder/FreeCAD #196: Sketcher - diameter sign alignment

xplwowi, 2021-10-13:
Hi. I took a little boldness to bother you a bit...
This is the problem coming from upstream that apeared in FreeCAD yet at the end of 2020. At that time, changes were made to UI scaling - auto factors were introduced. I think it may relate to the specifics of my environment - Linux, KDE Plasma laptop with 4k screen.
Diameter dimension label preceded with QChar(8960) (in ViewProviderSketch.cpp) is not properly aligned:
[image]

Everything looks fine when I use "empty set symbol" - QChar(8709):
[image]

I cannot find a piece of code in which there may be such a shift in the position of the label.

Environment:
OS: Ubuntu 21.04 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.28572 (Git)
Build type: Unknown
Branch: LinkDaily
Hash: fcb79d874627103151e95209819a5166d4f37254
Python version: 3.9.5
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.2
Locale: Polish/Poland (pl_PL)

luzpaz, 2022-01-18:
@xplwowi please retest LinkDaily to see if this still persists

xplwowi, 2022-01-19:
@luzpaz nothing has changed
[image]

Meanwhile the problem was resolved upstream
[image]

xplwowi, 2022-01-19:
Aparently, they did more or less the same what I do everytime when RT touches ViewProviderSketch.cpp: ad408a0989e90c9a1bc19337b29a8b4ec0f6c611
@realthunder could you be so kind to merge this small fix?