realthunder/FreeCAD #202: Tricky fillet results in a crash (but only with a particular selection order)

unrelentingtech, 2021-11-05:
- LinkDaily 5310be6aeff63255157e39f495a8fe3d82ec9243
- Test file: [TestCaseCrashNCollection_BaseSequence_Find.zip](https://github.com/realthunder/FreeCAD/files/7487191/TestCaseCrashNCollection_BaseSequence_Find.zip) (rename `.zip` to `.FCStd`)

Doing this kind of fillet:

[image]

on each side individually works fine, but fails as one operation in two different ways.

Starting on the "rear" side and adding the "front" side: "recompute failed":

[image]

But starting on the "front" side and then clicking the "rear" edge crashes!

```
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV
  * frame #0: 0x000000085d964d7e libTKFillet.so.7`ChFi3d_StripeEdgeInter(opencascade::handle<ChFiDS_Stripe> const&, opencascade::handle<ChFiDS_Stripe> const&, TopOpeBRepDS_DataStructure&, double) + 382
    frame #1: 0x000000085d950232 libTKFillet.so.7`ChFi3d_Builder::Compute() + 3058
    frame #2: 0x000000085dae03ca libTKFillet.so.7`BRepFilletAPI_MakeFillet::Build() + 42
    frame #3: 0x0000000868686dcd libTKTopAlgo.so.7`BRepBuilderAPI_MakeShape::Shape() + 29
    frame #4: 0x000000085d71d8fb Part.so`Part::TopoShape::makEShape(BRepBuilderAPI_MakeShape&, Part::TopoShape const&, char const*) + 107
    frame #5: 0x000000085d73230f Part.so`Part::TopoShape::makEFillet(Part::TopoShape const&, std::__1::vector<Part::TopoShape, std::__1::allocator<Part::TopoShape> > const&, double, double, char const*) + 367
    frame #6: 0x000000090cea2fb0 _PartDesign.so`PartDesign::Fillet::execute() + 704
    frame #7: 0x0000000807c335b5 libFreeCADApp.so`App::DocumentObject::recompute() + 85
    frame #8: 0x000000085d4d938c Part.so`Part::Feature::recompute() + 12
    frame #9: 0x000000090ce163a7 _PartDesign.so`PartDesign::Feature::recompute() + 279
    frame #10: 0x0000000807b97def libFreeCADApp.so`App::Document::_recomputeFeature(App::DocumentObject*) + 111
    frame #11: 0x0000000807b9a763 libFreeCADApp.so`App::Document::recomputeFeature(App::DocumentObject*, bool) + 339
    frame #12: 0x0000000a88335fc7 PartDesignGui.so`PartDesignGui::TaskDressUpParameters::syncItems(std::__1::vector<App::SubObjectT, std::__1::allocator<App::SubObjectT> > const&, bool) + 1815
    frame #13: 0x0000000a8833514b PartDesignGui.so`PartDesignGui::TaskDressUpParameters::onSelectionChanged(Gui::SelectionChanges const&) +1451
    frame #14: 0x00000008063e48fc libFreeCADGui.so`Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) + 28
…
```

unrelentingtech, 2022-02-11:
Still happens with OpenCASCADE 7.6.0 (and LinkDaily 91c5e00f6fd).

Shkolik, 2022-07-15:
Still somewhat happens... I created more simple test file, hope it will help.

[fillet-fail.zip](https://github.com/realthunder/FreeCAD/files/9118030/fillet-fail.zip)

Fillet on a right side working fine no difference what edge was selected, curved or strait.

Fillet on a left side will be created distorted if strait edge selected (but will be fine if curved edge selected):
[image]

When trying to crate fillet on both sides shape became invalid:
[image]

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git)
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United States (en_US)